RMerl/asuswrt-merlin.ng #459: RT-AC88U CPU1+2 go from 0%-100% load

lobocobra, 2020-03-23:
**Issue description:**
CPU load of CPU1&2 alternate from 0% to 100% to 0%. While one CPU is at 0% the other is at 100%. This goes on for 1-2hours and then eventually stops. 
=> See also issue #163. 
=> The device is heating up strongly compared to any other device, that was on same spot.
=> I have during such situation ping losses and connection issues with stall of connection.

There seem to be no relation between any activity I do in the house and this behavior. I assume it has to do with some internal process going mad. During such phases connection becomes unreliable

My Old Archer C9 which was before there had never more than 30% of load and this will many features activated. 

**My setup:**
1x RT-AC88U with Merlin 384.15 (aimesh host) all connected with backhaul.
1x RT-68U (aimesh node) 385.20253
2x Lyra (aimesh node) Original FW 384.46249
- I have currently no scripts or other services running. I am in the installation phase and try to stabilize the system before I do anything.
- No VPN, nothing running. Just a plain setup after factory reset.
- No FW rules only DMZ activated

Yesterday I was near to throw all the asus to the trash bin and re-install my old devices. Then I reset to factory each single device and one of my switch indicated a loop from a Lyra. After reset, this seems to have stabilized, but Ping loss still happen during those behaviors.

Anyone an idea?

lobocobra, 2020-03-23:
[image]

I know that it would be better to have all cables going to one central place, but cascading switches should not represent an issue, as no cable is longer that 150m. I have CAT7a end2end cabling.
=> With the old setup with some old Archer I had never ever one issue (except when I walked around the house :-) )

lobocobra, 2020-03-24:
I start now to downgrade the firmware to see if this behavior dis-appears....

lobocobra, 2020-03-24:
I installed stockfirmware FW_RT_AC88U_300438481116:

**All issues are now solved:**
- CPU load does not fluctuate any longer and is stable at 1-2%
- AIMesh over a SWITCH is again possible and does not create a loops. Previously I was not able to make AIMESH when a SWITCH was in between. Now it works out of the box. I had connection interruption and from 4 pings 1-2 arrived. Now is all solved.

Merlin is a great project, but there seems to be a major bug in the AIMESH part. IF you use backhaul over a switch, connection becomes unreliable and the load goes mad. My managed nod reported by blinking a LOOP, which is not explainable for me. After I have the stockfirmware and the devices are connected the same way. all works again.

Syst3mSh0ck, 2020-03-28:
AiMesh isn't supported in Merlin. It's closed-source so any changes made in this custom firmware could impact the functionality of AiMesh.

LuisSerrano, 2020-03-30:
384.13 (31-July-2019)
  - NEW: **AiMesh Router and node support**.  Note that automatic live
         update of Merlin-based nodes is not supported, you will have
         to manually update any Merlin-based nodes when a new firmware
         is available.  Asus-based nodes (which is recommended) will be
         able to make use of the automatic live update.

lobocobra, 2020-03-30:
Thanks for the responses. While moving around with mobile phones works now with AiMesh (with the original Firmware), it seems that the SW has problems with static clients (shelly, etc).

I was able to make it stable on Lyra/68U but on the main Node 88U, clients are rejected regularily.
=> So either I have to setup again AP (why the heck did I remove my dd-WRT devices)
=> Or I live with my current setup, to have an old AP next to the 88U to have the 1st floor covered

It seems to me that AiMesh is a nice idea, working for some users with 10 devices, but if you have >60 clients in a house the system does not work properly.

For sure not an issue for Merlin. I close the topic, maybe it helps someone else.

closed 2020-03-30 by lobocobra

LuisSerrano, 2020-03-31:
But the cpu loads with the Merlin software it's very high, 
I have switched to ASUS firmware , I have IMESH configured and the CPU load is normal

lobocobra, 2020-04-01:
I think Syst3mSh0ck brought it to the point. The AImesh part is closed-source-code and the team here can not know, what happens in there. Specially as the AImesh is still under heavy development. 

=> I am now in contact with the ASUS R&D and they created a beta for me to test in my environement, which already improved my connection issues with AImesh. 

The conclusion is, that if you need AImesh then it might work with Merlin or NOT.
=> My impression is, that the issue I described above is not due to Merlin but to errors in the closed code and they are working on it.