RMerl/asuswrt-merlin.ng #460: 384.15 - Apostrophe is not escaped in USB Device name resulting in Javascript error

cowmix88, 2020-03-24:
[image]

My USB device has an apostrophe in the device name for some odd reason which causing a JS error and preventing me from loading my usb device in the web ui folder list for usb applications. This is preventing me from selecting shared folders.

/aidisk/getfolderarray.asp

```html
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html>
<head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8" />
<meta HTTP-EQUIV="Pragma" CONTENT="no-cache">
<meta HTTP-EQUIV="Expires" CONTENT="-1">
<link rel="shortcut icon" href="images/favicon.png">
<link rel="icon" href="images/favicon.png">
<script>
var result = ['SanDisk' Cruzer Fit#0#1'];
var layer_order = '0';
var motion = 'gettree';
function get_folder_tree_success(){
parent.get_tree_items(result, motion);
}
</script>
</head>
<body onload="get_folder_tree_success();">
</body>
</html>
```

> **getfolderarray.asp:10 Uncaught SyntaxError: Unexpected identifier
getfolderarray.asp:18 Uncaught ReferenceError: get_folder_tree_success is not defined
    at onload (getfolderarray.asp:18)
onload @ getfolderarray.asp:18**

cowmix88, 2020-03-24:
Is there a way to possible force the name for the USB device and not use the one provided by the chip on the device?

Syst3mSh0ck, 2020-03-28:
This is a bug, but I would just remove the character in the Volume Label that's causing the problem for now.

RMerl, 2020-03-29:
Should be fixed with af5c1e32c1ee25eb346d38444555706b86b962d4 .

closed 2020-03-29 by RMerl