RMerl/asuswrt-merlin.ng #465: [Feature Request] Cloudflare for Famlies

peepsnet, 2020-04-02:
https://blog.cloudflare.com/introducing-1-1-1-1-for-families/

Can we add these to WAN DNS Page and DNSFilters dropdowns??

RMerl, 2020-04-02:
Will be eventually considered, but right now it's not possible, their DoT support for it is not yet functional according to their blog.

peepsnet, 2020-04-02:
At least the DNSFilter page... Those are not DOT right now

RMerl, 2020-04-02:
Undecided yet.  I want to avoid overcrowding this list with every single service available (there are countless ones available by now).  Would have to wait at least a few months to see if their service gains enough popularity and proves to be reliable enough to be worth adding to DNSFilter.

It's more likely to make it to the DNSPrivacy list since I already have regular 1.1.1.1 in there.

dave14305, 2020-04-04:
Until they allow customization of the family filters (1.1.1.3), it shouldn’t be rushed into the firmware. The malware list (1.1.1.2) may be useful as-is, like Quad9 is. I’m still concerned about their REFUSED responses.

RMerl, 2020-04-04:
I expect Cloudflare to know their stuff when it comes to DNS, so I'd like for them to explain us the rationale behind that decision.  Because from what I can find in RFCs, REFUSED is typically to denote a refusal to handle a client request based on policy.  That would probably imply that a resolver might then retry that same query with a secondary nameserver, which sounds quite wasteful.

dave14305, 2020-04-04:
Interesting that now I see it returning 0.0.0.0, which will now lead to Rebinding issues, but I guess it doesn’t matter for something you want blocked.
`dig phishing.testcategory.com @1.1.1.2

; <<>> DiG 9.14.8 <<>> phishing.testcategory.com @1.1.1.2
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 10211
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1452
;; QUESTION SECTION:
;phishing.testcategory.com.     IN      A

;; ANSWER SECTION:
phishing.testcategory.com. 60   IN      A       0.0.0.0

;; Query time: 12 msec
;; SERVER: 1.1.1.2#53(1.1.1.2)
;; WHEN: Sat Apr 04 14:49:27 DST 2020
;; MSG SIZE  rcvd: 95 `

closed 2021-01-03 by RMerl