RMerl/asuswrt-merlin.ng #467: Can't SSH into Asus RT-AC66U_B1 router

TrackRunny, 2020-04-03:
Router version: 384.15

SSH Settings: 
* Enable SSH: Lan + WAN
* Allow SSH port forwarding: no
* SSH port: 22
* Allow password login: no
* Enable SSH brute force protection: yes
* Using a authorized key: yes
* idle timeout: 20 minutes

Problem:

Despite setting a public and private keypair for this router I cannot log in. I have tried to login to my my user that I use on the web interface like so: `ssh user@192.168.1.1` after so it says the following: `user@192.168.1.1: Permission denied (publickey).` I have also checked the logs file inside the router which shows this: `Apr  2 17:45:16 dropbear[1485]: Login attempt for nonexistent user from 192.168.1.100:42042`. I am not sure what the issue is with this router and I would like to know possible solutions to be able to SSH into the router.

RMerl, 2020-04-03:
Your problem is you're not specifying the correct user.  This has nothing to do with the firmware, and is not a bug.

closed 2020-04-03 by RMerl

TrackRunny, 2020-04-03:
However, I am sure I am specifying the correct user. I am trying to SSH into the router with the user I use for the web login page.