RMerl/asuswrt-merlin.ng #482: Any chance to update dropbear with Ed25519 support

lorenc-tomasz, 2020-04-10:
Currently support is added to master branch https://github.com/mkj/dropbear/pull/91 Do you consider adding it to FW?

RMerl, 2020-04-10:
Already added a while ago, and part of the 384.16 release.  Themiron actually used Asuswrt-Merlin for testing his implementation of it before pushing it on dropbear's repo.

closed 2020-04-10 by RMerl

lorenc-tomasz, 2020-04-10:
@RMerl But it's not a part of 384.13_6. So e.g. RT-AC87U users don't have it.

RMerl, 2020-04-10:
The RT-AC87U and RT-AC3200 are now considered to be legacy devices, and will mostly receive only bugfixes.

lorenc-tomasz, 2020-04-10:
Understand but I hope that you will find a spare time to add Ed25519 to 13_x version as edDSA is considered as more secure than RSA. It's always a good option to have possibility to use better security :) 

Thank you for the answer :)

lorenc-tomasz, 2020-04-13:
@themiron Will you consider adding to 13_x? :)

themiron, 2020-04-14:
> @themiron Will you consider adding to 13_x? :)

sure, when I have a time

lorenc-tomasz, 2020-04-23:
> > @themiron Will you consider adding to 13_x? :)
> 
> sure, when I have a time

@themiron As 13.7 is on the way, maybe you will find some time to add  Ed25519 :)

RMerl, 2020-04-23:
Now is not the time.  The current beta cycle will be very short, as the goal is to get OpenSSL security updates out as soon as possible.  So, no new feature should be added right now.

There is zero urgency in adding ed25519 support, this is just a "nice to have" feature.  Managing your own personal router within your LAN using the currently supported ciphers/KEX methods is perfectly fine.

themiron, 2020-04-30:
> @themiron Will you consider adding to 13_x? :)

done with be3f5f416810ca60388deeaa9138392095d558fd

lorenc-tomasz, 2020-04-30:
> > @themiron Will you consider adding to 13_x? :)
> 
> done with [f593d04](https://github.com/RMerl/asuswrt-merlin.ng/commit/f593d0440d454fcf22ce434c8b149a605dc617a2)

Thank you :))