RMerl/asuswrt-merlin.ng #485: Set custom DHCP client options on WAN

loicortola, 2020-04-13:
Hello,
first I want to say congrats for the awesome work around this firmware.
You guys rock!

I noticed more and more ISPs are moving away from PPPoE to DHCP and requiring custom DHCP client options.

My ISP (Orange France) for instance, requires Option 77 and Option 90 to be set to specific values.
I guess each ISP has various options and it would be weird as f*** to implement ISP-based recommendations.
However, would it be possible to have a UI dedicated to this?
I'm thinking two possibilities:
 - v0: just one textarea with JSON or Yaml or anything key/value pairs like : {"option-77": ".....", } etc...
 - v1: a clean interface allowing us to add specific options one by one (like the port range UI).

What do you guys think about this feature request?

If you think this might be interesting, I'd like to help with the implementation although my development skills are mostly Web (I could help with UI only)

Thanks and Happy easter :)

loicortola, 2020-05-24:
@RMerl @themiron , any update on this?
Thanks

RMerl, 2020-05-24:
Not in my short terms plans, sorry.

loicortola, 2020-05-25:
Are you guys open to Pull Requests?
I think I could help manage at least the web version

closed 2021-01-03 by RMerl