RMerl/asuswrt-merlin.ng #487: 384.16 Poor WiFi performance

chrisknowles13, 2020-04-14:
RT-AX88U upgraded to 384.16 and experienced significant and poor performance of wifi primarily devices on 5GHz band (didn't test on 2.4GHz) with devices experiencing constant drop outs, poor signal and slow/erratic  monitored speeds using internet speed tests.  After a few days reverted back to 384.15 and everything is working again.

asaddon, 2020-04-14:
Wifi issues are pretty hard to troubleshoot and you need to do extensive testings to find the cause and fix it. And in any case, all wifi related stuff is totally closed source and Merlin can't do anything to fix it so it's useless to open an issue here. You have to report it to Asus if you really think it's a bug in the firmware.

chrisknowles13, 2020-04-14:
Hi, thanks for the reply.  Apologies for posting on here, I read the part about using SNB forums after the fact.  I headed over there and looks like someone else is experiencing the same/similar issues with 384.16 on the RT-AX88U:

https://www.snbforums.com/threads/latency-wifi-5ghz-rt-ax88u.63318/

Thanks again, and great work

closed 2020-04-14 by chrisknowles13