RMerl/asuswrt-merlin.ng #488: ipv6 not work on 384.16

gxing, 2020-04-14:
I upgrade my RT-AC66-B1 from 384.15 to 383.16.
ipv6 setitng: native/Stateful.

Hosts in the LAN can get the IPv6 address.

But I can not ping the hosts in the lan from the internet.

Close/Open IPv6 firewall is useless.

When I rollback to 384.15, eveything works fine.

br

themiron, 2020-04-14:
please provide more info from
1. IPv6 configuration page
2. System Log / IPv6 page
IPv6 stops announcing if there's no working IPv6 from your ISP, can't exclude this case so far

gxing, 2020-04-15:
IPv6 page and IPV6 firewall page.
I already rollback to 384.15,so no syslog is saved.

BTW:
When I use ipv6 connect to the remote site, the remote site say my IPv6
address is the
router's address 2408:xxxx::1, not the host's IPv6 address 2408:xxxx::1xxx.
I remeber the old version can report the true IPv6 address to the remote
host.
This is what I see in the 384.15.

gxing, 2020-04-15:
attach the syslog(no cleanup)  and the dmesg(384.15)
I do not see useful info about the ipv6.
BTW: I upgrade to 384.16 and downgrade to 384.15 yesterday(Apr 14) about
5-6pm.

gxing, 2020-04-15:
do not konw why I reply by email, the attachment is missing.
Repost it on git hub.

[image]
[image]
[syslog.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/4478585/syslog.txt)
[dmesg.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/4478586/dmesg.txt)

RedLucas, 2020-04-27:
FWIW, I’m on the same version, with an r7000 and my ipv6 does work.

FrancisCTN, 2020-06-02:
I'm having the same problem too, on newer firmware 384.17 with my RT-AC5300, IPv6 is non-functional. Correct settings as per ISP instructions for IPv6 configuration page (Native, DHCP-PD, Release Prefix, Stateless, Auto DNS, Router Advertisement).

**Troubleshooting (So Far)**

**ping6** to www.google.com results in dnsmasq correctly responding to the query[AAAA] with www.google.com IPv6 address of 2404:6800:4006:80b::200e but there is no response after that point (100% packet loss).

**traceroute6** to www.google.com results in the first hop going to the router (Address is the same as the webUI Lan IPv6 Address Field), but hops after that point all fail / timeout...

**traceroute6** to IPv6 google DNS Server (2001:4860:4860::8888) has the same failure / timeout result as above.

I wonder if there is a misconfigured firewall, interface definition or route somewhere

closed 2021-01-03 by RMerl