RMerl/asuswrt-merlin.ng #490: Firmware successfully uploaded to AC66U_B1, will not boot afterwards.

leon332157, 2020-04-14:
I have tried to upload the RT-AC68U_384.16_0 firmware to my AC66U_B1, the admin panel shows successful upload. but after reboot the router goes into rescue mode and I had to restore to an ASUS provided firmware. Is this router still supported?

vladut123, 2020-04-14:
What you had done is wrong because you uploaded rt-ac68u to rt-ac66u_b1

leon332157, 2020-04-14:
Based on this (link)[https://github.com/RMerl/asuswrt-merlin.ng/wiki/Supported-Devices] I'm supposed to use the rt-ac68u software for a rt-66ub1

RMerl, 2020-04-14:
Could be a number of reasons, from downloading the wrong model to a browser-related issue with the firmware update process.  Please head to the support forums at SNBForums for further assistance, this tracker is intended for development-related issues.

closed 2020-04-14 by RMerl