RMerl/asuswrt-merlin.ng #498: Adaptive QoS doesn't work on RT-AX58U (384.16)

ilmakangas, 2020-04-22:
This feature can be enabled on UI without any errors in the system log. But no bandwidth is being limited for any application and on the QoS classification UI page no packet flow is being shown at all, all the rates stay at 0.

Changing any of the adaptive QoS settings including "Queue discipline" doesn't have any effect on this.

ilmakangas, 2020-04-22:
Seemed to start working after reflashing Merlin and restoring to factory settings before and after that. At first didn't work even with factory settings for some reason.

tekneek, 2020-04-24:
I have had the same experience. 

When troubleshooting it worked as expected (for me) w/ a factory reset install of the stock firmware ASUS RT-AX3000 Firmware version 3.0.0.4.384.8591. I don't know when those elements will be incorporated into Asuswrt-Merlin (or if they already are?).

RMerl, 2020-04-24:
Until Asus releases newer source code there is nothing I can do about it.

closed 2020-04-24 by RMerl