RMerl/asuswrt-merlin.ng #499: RT-AC68U 384.16 More than 4 hosts/alias for an ip address in /etc/hosts results in seg fault in openvpn client and ntpd

snaunton, 2020-04-24:
RT-AC68U 384.16 

The openvpn executables will segfault with more than 4 hosts/alias for an ip address in /etc/hosts and Log verbosity for the vpn client is set to 0 or 1.

ntpd will segfault with more than 4 hosts/aliases in /etc/hosts

To reproduce:

1) Add the following to /etc/hosts

192.168.1.1 host1 host2 host3 host4 host5

(Note you can use different ip addresses and hostnames/aliases)

2) Stop and start an openvpn client and/or ntpd. Note that the executables do not actually start:

ps | grep openvpn
ps | grep ntp

Openvpn client will show connecting status in the webgui.

/etc/openvpn/client? (where ? is the vpn client number) will have been created.

If you have strace installed (via entware probably), and the /etc/openvpn/client? exists, running (where ? is the vpn client number):

strace -f /etc/openvpn/vpnclient? --cd /etc/openvpn/client? --config config.ovpn

You will see:

open("/etc/hosts", O_RDONLY)            = 5
ioctl(5, TCGETS, 0xbee1b95c)            = -1 ENOTTY (Inappropriate ioctl for device)
read(5, "127.0.0.1 localhost.localdomain "..., 4096) = 727
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x6ba4ffb4} ---
+++ killed by SIGSEGV +++

Running ntpd/ntp under strace (strace /usr/sbin/ntp -t -S /sbin/ntpd_synced -p pool.ntp.org -l -I br0) has exactly the same result.

If you remove host5 all is well.

RMerl, 2020-04-24:
This is an old uclibc limitation.

snaunton, 2020-04-24:
Closing.
Summary:
Do no use any more than 4 hostnames or aliases in a line in /etc/hosts.

closed 2020-04-24 by snaunton