RMerl/asuswrt-merlin.ng #503: RT-AX88U SSID reverts on reboot

roblandry, 2020-04-27:
Current version: 384.17
Previous version: 384.16

AiMesh with:RT-AX58U and RT-AC68P all up to date with 384.17.

Updated yesterday. Daily reboot setup in gui.
Upon reboot, SSID reverts to the default ASUS and ASUS_5G SSIDs, auth goes to open (both), channels change to auto (both), 2g channel goes to 20mhz.

roblandry, 2020-04-27:
Rolled back to 384.16, reset several times, and settings will not stick. Each reboot resets router to factory now. Going to revert to stock and see if I can migrate back to merlin with successful settings being saved.

kissglay, 2020-04-27:
apparently faulty nvram, not firmware issue.

roblandry, 2020-04-27:
So seems to be saving now.
I rolled back to 384.16. 
  Setup basics. Upon restart, settings reverted.
  Hard reset, setup basics. Upon restart, settings reverted.
Installed latest stock FW.
  Setup basics. Upon restart settings were present.
  Hard reset, setup basics. Upon restart, settings were present.
Updated back to 384.17.
  Setup basics. Upon restart, settings were present.

I did notice, that on first upgrade, from 16 -> 17, a manual reboot was required following install (which i haven't seen before, it was GUI directed)
However, this time, a manual reboot was not required.

Im not sure if this was just a bug in my router or in the fw, I will happily close if it was my router.

closed 2021-01-03 by RMerl