RMerl/asuswrt-merlin.ng #507: Wrong brace in lighttpd-1.4.39/src/arpping_thread.c

odkrys, 2020-05-04:
https://github.com/RMerl/asuswrt-merlin.ng/blob/3194f1464bc5fedefe9a1bdb964c6ccb72b5100e/release/src/router/lighttpd-1.4.39/src/arpping_thread.c#L673

brace should be remove or closed.

```
gdb ./lighttpd-arpping
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./lighttpd-arpping...
(gdb) r -f br0
Starting program: /tmp/mnt/ssd/movie/lighttpd-arpping -f br0
warning: Unable to determine the number of hardware watchpoints available.
warning: Unable to determine the number of hardware breakpoints available.
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.

Program received signal SIGSEGV, Segmentation fault.
0x00013848 in dumparptable () at arpping_thread.c:695
695                                     if( ( strcmp(p->ip->ptr, ip_entry)!=0 && strcmp(p->mac->ptr, hw_address)==0 ) ||
```

RMerl, 2020-05-05:
Removing/closing that brace means you would read the whole file, but only process the very last line, which makes no sense to me.

odkrys, 2020-05-06:
sorry. I forgot to change language to C in notepad :O.
I replaced the code with the following and it worked fine.

```
				if( ( strcmp(p->ip->ptr, ip_entry)!=0 && 
					/*strcmp(p->mac->ptr, hw_address)==0 )*/ ||
					 strcmp(p->mac->ptr, "00:00:00:00:00:00")==0 ){
```
or

```
	char ip_entry[32], hw_type[8], flags[8], hw_address[32], mask[32], device[8];
```

hw_address[32] -> hw_address[64]

I got an unhandled level 3 translation fault (11) on every reboot of the RT-AC86U, which caused lighttpd-arpping not to run.

odkrys, 2020-05-06:
It seems to be fine in official merlin.
I will find out more.

odkrys, 2020-05-06:
I reproduced it.
I don't know why, but I definitely remember seeing the address 00:00:00:00:00:00 in cat /proc/net/arp result when debugging with gdb.

So I manually added that address.
```
arp -s 192.168.50.2 00:00:00:00:00:00
```
And it happened.
```
kernel: lighttpd-arppin[15653]: unhandled level 3 translation fault (11) at 0x00000000, esr 0x92000007
kernel: pgd = ffffffc01577f000
kernel: [00000000] *pgd=00000000132cf003, *pud=00000000132cf003, *pmd=000000001118c003, *pte=0000000000000000
kernel: CPU: 0 PID: 15653 Comm: lighttpd-arppin Tainted: P           O    4.1.27 #32
kernel: Hardware name: Broadcom-v8A (DT)
kernel: task: ffffffc0191ae100 ti: ffffffc0112b4000 task.ti: ffffffc0112b4000
kernel: PC is at 0x13848
kernel: LR is at 0x13814
kernel: pc : [<0000000000013848>] lr : [<0000000000013814>] pstate: 20010010
kernel: sp : 00000000ffd3e7a0
kernel: x12: 000000000002625c 
kernel: x11: 0000000000000000 x10: 000000000022d748 
kernel: x9 : 0000000000028e40 x8 : 0000000000231538 
kernel: x7 : 00000000ffd3e7e8 x6 : 0000000000028e40 
kernel: x5 : 0000000000000000 x4 : 000000000022d6d0 
kernel: x3 : 0000000000000031 x2 : 0000000000000039 
kernel: x1 : 00000000ffd3e7e8 x0 : 0000000000000008 
```

I'm not sure why the arp table had the address 00:00:00:00:00:00, but I've had this problem for more than a few months and I've been tracking it a few days ago.

odkrys, 2020-05-06:
After I did nvram unset webdav_smb_pc, all looks fine. I don't know exactly well..

odkrys, 2020-05-06:
It happened again after reboot.
What I suspect is the wrong format of webdav_smb_pc.
```
DESKTOP-68E357Q<192.168.50.81<aa:bb:cc:dd:ee:ff<1DESKTOP-68E357Q<192.168.50.158<11:22:33:44:55:66<
```

It is a record of connecting wifi and lan on the same PC.
Name ip mac online.
DESKTOP-68E357Q<192.168.50.81<aa:bb:cc:dd:ee:ff<1

https://github.com/RMerl/asuswrt-merlin.ng/blob/3194f1464bc5fedefe9a1bdb964c6ccb72b5100e/release/src/router/lighttpd-1.4.39/src/arpping_thread.c#L179

It seems that the first and second ones are not properly separated by the < mark.
aa:bb:cc:dd:ee:ff<1DESKTOP-68E357Q must be separated by aa:bb:cc:dd:ee:ff<1<DESKTOP-68E357Q.

```
nvram get webdav_smb_pc
DESKTOP-68E357Q<192.168.50.81<f2:52:71:b9:1e:d5<1DESKTOP-68E357Q<192.168.50.158<98:83:89:39:cc:55<
@RT-AC86U-69B8:/tmp/home/root# rm /tmp/lighttpd/lighttpd-arpping.pid
@RT-AC86U-69B8:/tmp/home/root# /usr/sbin/lighttpd-arpping -f br0
Segmentation fault
```
```
nvram set webdav_smb_pc="DESKTOP-68E357Q<192.168.50.81<f2:52:71:b9:1e:d5<1<DESKTOP-68E357Q<192.168.50.158<98:83:89:39:cc:55<"
@RT-AC86U-69B8:/tmp/home/root# rm /tmp/lighttpd/lighttpd-arpping.pid
@RT-AC86U-69B8:/tmp/home/root# /usr/sbin/lighttpd-arpping -f br0
```
it is running fine.

odkrys, 2020-05-06:
It happened again after reboot.
What I suspect is the wrong format of webdav_smb_pc.
```
DESKTOP-68E357Q<192.168.50.81<aa:bb:cc:dd:ee:ff<1DESKTOP-68E357Q<192.168.50.158<11:22:33:44:55:66<
```

It is a record of connecting wifi and lan on the same PC.
Name ip mac online.
DESKTOP-68E357Q<192.168.50.81<aa:bb:cc:dd:ee:ff<1

https://github.com/RMerl/asuswrt-merlin.ng/blob/3194f1464bc5fedefe9a1bdb964c6ccb72b5100e/release/src/router/lighttpd-1.4.39/src/arpping_thread.c#L179

It seems that the first and second ones are not properly separated by the < mark.
aa:bb:cc:dd:ee:ff<1DESKTOP-68E357Q must be separated by aa:bb:cc:dd:ee:ff<1<DESKTOP-68E357Q.

```
nvram get webdav_smb_pc
DESKTOP-68E357Q<192.168.50.81<aa:bb:cc:dd:ee:ff<1DESKTOP-68E357Q<192.168.50.158<11:22:33:44:55:66<
@RT-AC86U-69B8:/tmp/home/root# rm /tmp/lighttpd/lighttpd-arpping.pid
@RT-AC86U-69B8:/tmp/home/root# /usr/sbin/lighttpd-arpping -f br0
Segmentation fault
```
```
nvram set webdav_smb_pc="DESKTOP-68E357Q<192.168.50.81<aa:bb:cc:dd:ee:ff<1<DESKTOP-68E357Q<192.168.50.158<11:22:33:44:55:66<"
@RT-AC86U-69B8:/tmp/home/root# rm /tmp/lighttpd/lighttpd-arpping.pid
@RT-AC86U-69B8:/tmp/home/root# /usr/sbin/lighttpd-arpping -f br0
```
it is running fine.

odkrys, 2020-05-06:
I believe this is the right fix.
https://github.com/odkrys/asuswrt-merlin.ng/commit/26b41397d559d9e3a310e9bea7b3edc341152e74

closed 2021-01-03 by RMerl