RMerl/asuswrt-merlin.ng #512: Dead link in QoS - WAN/LAN Bandwidth Monitor

feranick, 2020-05-08:
Using 384.17. The URL link for WAN/LAN Bandwidth Monitor FAQ in the QoS - WAN/LAN Bandwidth Monitor (pointing to https://www.asus.com/support/FAQ/1010934) seems no longer existing. Possibly replaced by the URL: https://www.asus.com/support/faq/1008717/

RMerl, 2020-05-08:
Will be fixed eventually in GPL merges - Asus just keep changing the stupid URLs all the time...  I vaguely remember that specific URL having been fixed already on the ax branch following the 8543 merge, so it will be fixed whenever Asus starts releasing new GPLs for the mainline branch as well.

feranick, 2020-05-08:
Thanks. Not that the new link is perfect. All images are missing in the ASUS page...

borzaka, 2020-05-25:
I have RT-AC86U with the current latest official firmware: 3.0.0.4.384.81858
And it links to the right URL: https://www.asus.com/support/FAQ/1008717
No dead link.

RMerl, 2020-05-25:
My answer still stands.

feranick, 2020-06-21:
This is fixed in 384.18_beta1. I assume this bug report can be closed.

closed 2020-06-23 by RMerl