RMerl/asuswrt-merlin.ng #522: No "Access Intranet" option for guest network

johny-mnemonic, 2020-05-20:
I am on 384.17 and router RT-AC68U.
I have never tried setting up guest network before, so I may be doing something wrong, but I just went into "Guest Network" menu, and pushed the first out of three "Enable" buttons there for 5GHz.
I have set SSID and WPA2 and pushed "Apply" button.

New network appeared and I can connect to it. But to my surprise, I am still able to access my local network :-O So I started googling and found posts from people showing that there should be "Access Intranet" option in the UI, while I have no such option there.
Is there something I have to do elsewhere in the router to enable this?

RMerl, 2020-05-20:
Disable the option "Access Intranet" on your Guest.

RMerl, 2020-05-20:
[image]

closed 2020-05-20 by RMerl

johny-mnemonic, 2020-05-21:
If only I had that option, I would do it, but it is not there.

RMerl, 2020-05-21:
That is because you are running in AP mode, not in Router mode.  in AP mode, your RT-AC68U is only a bridge to your network, with no routing control, therefore the option is not available as only the primary router is able to handle that kind of functionality.

johny-mnemonic, 2020-05-21:
I see. That's sad. Asus firmware limitation I guess. Right?
So the only way is to find/create user script for that or find another compatible FW?

RMerl, 2020-05-21:
Not Asus firmware, it's a technical limitation.  An AP has no control over a separate Router,  The blocking you want need to be handled by the router, not by the AP.

johny-mnemonic, 2020-05-21:
In case the guest network was implemented as NAT no matter whether the box is in router or AP mode, it would be simple to prevent those guest clients to access intranet network apart from main router IP. This solution would be good enough to most I guess....