RMerl/asuswrt-merlin.ng #524: Segmentfault for all the shell commands except those from entware

ruoshan, 2020-05-25:
Hi, I'm having a weird issue that's hard to repro, but I get lucky this time when I'm still on the ssh.
I don't know what else I can do, I installed `strace` previously, so I try to strace all the commands like `ls` or `sh` (I guess they all go to busybox). Here is what I get:
```
admin@AC68U:/jffs# strace sh
execve("/bin/sh", ["sh"], 0xbe843d90 /* 14 vars */) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x400d4000
open("/lib/libcrypt.so.0", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=13195, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x400e2000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\4\0\0004\0\0\0"..., 4096) = 4096
mmap2(NULL, 118784, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x4018d000
mmap2(0x4018d000, 8948, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x4018d000
mmap2(0x40197000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2000) = 0x40197000
mmap2(0x40198000, 70248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x40198000
close(3)                                = 0
munmap(0x400e2000, 4096)                = 0
open("/lib/libm.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=68623, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x400ca000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\34#\0\0004\0\0\0"..., 4096) = 4096
mmap2(NULL, 102400, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x400d5000
mmap2(0x400d5000, 61388, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x400d5000
mmap2(0x400ec000, 4100, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xf000) = 0x400ec000
close(3)                                = 0
munmap(0x400ca000, 4096)                = 0
open("/lib/libgcc_s.so.1", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=41040, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x40033000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\260'\0\0004\0\0\0"..., 4096) = 4096
mmap2(NULL, 73728, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x40111000
mmap2(0x40111000, 39128, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x40111000
mmap2(0x40122000, 2800, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x9000) = 0x40122000
close(3)                                = 0
munmap(0x40033000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=440575, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x400f3000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\220I\1\0004\0\0\0"..., 4096) = 4096
mmap2(NULL, 475136, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x401aa000
mmap2(0x401aa000, 413376, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x401aa000
mmap2(0x40217000, 4980, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x65000) = 0x40217000
mmap2(0x40219000, 17872, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x40219000
close(3)                                = 0
munmap(0x400f3000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=440575, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=440575, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0555, st_size=440575, ...}) = 0
close(3)                                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0555, st_size=21388, ...}) = 0
open("/dev/urandom", O_RDONLY)          = 3
read(3, "\257\312\336\260", 4)          = 4
close(3)                                = 0
mprotect(0x40197000, 4096, PROT_READ)   = 0
mprotect(0x400ec000, 4096, PROT_READ)   = 0
mprotect(0x40217000, 4096, PROT_READ)   = 0
mprotect(0x400b9000, 4096, PROT_READ)   = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xaf9fb78} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

themiron, 2020-05-25:
please clarify your model and entware arch?

ruoshan, 2020-05-25:
it's AC68U on v384.17.
@themiron  Is there anything I can do to assist the debugging before I reboot? I need to reboot so I can see the entware version. I don't know if it can be reproduced.

ruoshan, 2020-05-25:
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=7.3.0
gcc_flags=-O2 -pipe -march=armv7-a -mtune=cortex-a9 -fno-caller-saves -mfloat-abi=soft
libc=glibc
libc_version=2.23

themiron, 2020-05-26:
perhaps it's memory issue, try to watch ram consumption

ruoshan, 2020-05-27:
> perhaps it's memory issue, try to watch ram consumption.
> or, try to reinstall current entware from the scratch

Nah, it's not the memory. I managed to login again by installing an extra dropbear using entware. the mem used was only at 50mb or so, the device has 256mb mem.

The SEGV is in the busybox of the ROM not from the entware, I don't know if it will have any relationship with entware.

themiron, 2020-05-27:
can you try to install gdb from entware and run sh under it?
$ gdb /bin/sh
$ r
wait for crash and issue following command:
$ bt full

ruoshan, 2020-05-28:
Yep, that's what's I was trying to do, but the `opkg install gdb` segfault too. 😞 

I reset the router, reinstall the entware last night. still have the same issue, but I can't ssh. I will post the backtrace once I hit it again.

The dropbear use `/bin/sh` when login, and it segfaults. I have to find a way to have dropbear use the `/opt/bin/sh` instead.

raven-au, 2020-05-31:
What do you mean "reinstall" the router, factory defaults reset and minimal config to check if that worked first perhaps?
What is your environment, eg. PATH, is there an LD_LIBRARY_PATH defined in there too?

closed 2021-01-03 by RMerl