RMerl/asuswrt-merlin.ng #532: Adaptive-QoS - Classification bug

CTOUT, 2020-06-04:
Hi,

Running 384.17 on an RT-AX88U
I've noticed an oddity in adaptive-QoS.  

If I have the configuration set to Custom, in the order:- 
Gaming
VoIP and IM
Video/Audio Streaming
Web Surfing
File Transfer
Others

Under the Classification tab, I correctly see "Microsoft Teams" being coloured as "VoIP and IM" but when hovering over the icon it states "Video and Audio", and "Battle.net" being coloured/listed as Gaming.

If I have the configuration set to Custom, in the order:- 
VoIP and IM
Gaming
Video/Audio Streaming
Web Surfing
File Transfer
Others

Under the Classification tab, I now see "Microsoft Teams" being coloured as VOIP listed (on hover) as Gaming, and "Battle.net" being coloured as Gaming but listed as "Video and Audio Streaming".

It may just be a UI glitch.  However, I don't really know if what I am seeing is what I am getting as a result.

RMerl, 2020-06-04:
I fixed something related to this in 384.18, however I didn't check if it was an old bug that previously existed, or was new to 384.18.  You can try with the latest alpha build to see if the issue is gone.

https://www.asuswrt-merlin.net/test-builds/

CTOUT, 2020-06-05:
Thanks @RMerl - I'll wait for that to become GA and give it a whirl then. 👍

closed 2021-01-03 by RMerl