RMerl/asuswrt-merlin.ng #533: When Smart Connect Enabled Wi-Fi Status shoes 2.4G & 5G Disabled

helsan74, 2020-06-04:
[https://imgur.com/a/QGMsYVE](url)
Is this normal? Thanks

Router: AX88u
FW: 384.17

RMerl, 2020-06-13:
Cannot reproduce here, works for me.  Make sure you didn't disable the radios through the button, the scheduler, or the enable/disable option.

helsan74, 2020-06-13:
I just validated all, and all good. WIFI also working great. Just shows Disabled.
In fact I have over 12 devices running in wifi without any issue.
Only thing I remember this started showing after I switch from Smart Connect to normal.

helsan74, 2020-07-03:
Was able to find the issue after upgrading to 384.18

under 384.17 under WIRELESS and PROFESSIONAL show enabled RADIO ON but they are disabled in reality.
under 384.18 under WIRELESS and PROFESSIONAL show enabled RADIO OFF and wireless doesnt work.

All need to fix the issues is to switch between to OFF reboot and then back to ON and reboot... all works again.

closed 2021-01-03 by RMerl