RMerl/asuswrt-merlin.ng #534: Clients: 0 (Searching, please wait...) Bug

slalom-ms, 2020-06-08:
Since the recent update 384.17 on my AC68U The Clients tab on the front page shows 0 and just shows "Searching, please wait...". I reset the router and all the settings and its still the same issue. Any idea?

[image]

[image]

asaddon, 2020-06-09:
Is this a genuine AC68U of a transformed T-Mobile unit?

asaddon, 2020-06-09:
Or*

slalom-ms, 2020-06-09:
Transformed T-Mobile unit. But it was working on previous updates.

RMerl, 2020-06-09:
This model is not supported, and never was either.  Asus has taken active measures over time to prevent people from actively hacking it, and one of the symptoms of their measures is exactly what you describe - the networkmap not working.

closed 2020-06-09 by RMerl