RMerl/asuswrt-merlin.ng #535: wrong dns servers reported by ipv6 router advertisement

dns53, 2020-06-08:
I have ipv6 set up and have configured custom dns servers but this is not being sent in the router advertisement.
Instead only the router is reporting itself as the dns server.

I have configured the following settings:
IPV6 > IPv6 DNS Setting
Connect to DNS Server automatically:  Disable
IPv6 DNS Server 1: configured with a ipv6 address of a dns server
IPv6 DNS Server 2: configured with a ipv6 address of a dns server
IPv6 DNS Server 3: configured with a ipv6 address of a dns server

Running radvdump i am receiving the following.
I am expecting my configured servers to to be in the RDNSS section.

interface eth0
{
        AdvSendAdvert on;
        # Note: {Min,Max}RtrAdvInterval cannot be obtained with radvdump
        AdvManagedFlag off;
        AdvOtherConfigFlag on;
        AdvReachableTime 0;
        AdvRetransTimer 0;
        AdvCurHopLimit 64;
        AdvDefaultLifetime 600;
        AdvHomeAgentFlag off;
        AdvDefaultPreference medium;
        AdvLinkMTU 1492;
        AdvSourceLLAddress on;

        prefix 2001:44b8:xxxx:xxxx::/64
        {
                AdvValidLifetime 600;
                AdvPreferredLifetime 600;
                AdvOnLink on;
                AdvAutonomous on;
                AdvRouterAddr off;
        }; # End of prefix definition

        DNSSL home
        {
                AdvDNSSLLifetime 600;
        }; # End of DNSSL definition

        RDNSS 2001:44b8:xxxx:xxxx::1
        {
                AdvRDNSSLifetime 600;
        }; # End of RDNSS definition

}; # End of interface definition

themiron, 2020-06-08:
this is desired behavior. why you need to stick with external dns on clients?

dns53, 2020-06-09:
The main reason is i have an internal .home to resolve internal applications such as home automation, download managers, git servers etc.

dnsmasq is configured to forward requests to another server but this caches requests for a long time.
When I make changes they seem to get stuck for a longer period of time until I restart the router to flush the cache.
If I can specify my dns servers this will point to internal computers that have a full copy of the home zone so gets up to date results when I modify a domain name.

I run 3 dns servers on the router, dnsmasq to forward to bind and unbound.
If I could stop dnsmasq and have bind running on port 53 it would work better but I don't think that is possible.

themiron, 2020-06-09:
eyerything is possible, check out http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html
dnsmasq supports various ttl settings, as well as changing port.

closed 2020-06-09 by themiron