RMerl/asuswrt-merlin.ng #536: RT-AX5xU - wl0 crashes with fatal error

Wri84, 2020-06-08:
as discussed [here](https://www.snbforums.com/threads/rt-ax56u-wifi-2-4ghz-disconnected-wl0-fatal-error.63642) the 2.4 ghz wl0 interface crashes with fatal errors and doesn't come up under Merlin 384.17 stable as well as 384.18 alpha 1.

OP writes about RT-AX56U, but same problem occurs on my brandnew RT-AX58U. Hardware reset via SSH "nvram erase" done, no change.

[Settings_RT-AX58U.zip](https://github.com/RMerl/asuswrt-merlin.ng/files/4748958/Settings_RT-AX58U.zip)
> Jun  9 00:32:06 kernel: wl0: fatal error, reinitializing, total count of reinit's[2]
Jun  9 00:35:27 kernel: wl0: fatal error, reinitializing, total count of reinit's[3]
Jun  9 00:38:48 kernel: wl0: fatal error, reinitializing, total count of reinit's[4]
Jun  9 00:42:09 kernel: wl0: fatal error, reinitializing, total count of reinit's[5]

RMerl, 2020-06-09:
Everything tied to wifi is closed source, and outside of my control.

closed 2020-06-09 by RMerl

paldier, 2020-06-09:
You have to perform a "factory resetting"
Erasing nvram is not equivalent to factory resetting