RMerl/asuswrt-merlin.ng #539: asus rt-ax88u random reboots

jepan, 2020-06-21:
asus rt-ax88u random reboots on merlin 384.17

jepan, 2020-06-21:
Here’s the log:
May  5 13:05:05 syslogd started: BusyBox v1.25.1
May  5 13:05:05 kernel: klogd started: BusyBox v1.25.1 (2020-04-25 22:29:01 EDT)
May  5 13:05:05 kernel: Booting Linux on physical CPU 0x0
May  5 13:05:05 kernel: Linux version 4.1.51 (merlin@ubuntu-dev) (gcc version 5.5.0 (Buildroot 2017.11.1) ) #4 SMP PREEMPT Sun Apr 26 00:24:46 EDT 2020
May  5 13:05:05 kernel: CPU: AArch64 Processor [420f1000] revision 0
May  5 13:05:05 kernel: Detected VIPT I-cache on CPU0
May  5 13:05:05 kernel: alternatives: enabling workaround for ARM erratum 845719
May  5 13:05:05 kernel: PERCPU: Embedded 15 pages/cpu @ffffffc03ffb4000 s21400 r8192 d31848 u61440
May  5 13:05:05 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 236845
May  5 13:05:05 kernel: Kernel command line: coherent_pool=4M cpuidle_sysfs_switch pci=pcie_bus_safe rootwait
May  5 13:05:05 kernel: PID hash table entries: 4096 (order: 3, 32768 bytes)
May  5 13:05:05 kernel: Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
May  5 13:05:05 kernel: Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
May  5 13:05:05 kernel: Memory: 903344K/960512K available (4822K kernel code, 812K rwdata, 1624K rodata, 228K init, 532K bss, 57168K reserved, 0K cma-reserved)
May  5 13:05:05 kernel: Virtual kernel memory layout:
May  5 13:05:05 kernel:     vmalloc : 0xffffff8000000000 - 0xffffffbdffff0000   (   247 GB)
May  5 13:05:05 kernel:     vmemmap : 0xffffffbe00000000 - 0xffffffbfc0000000   (     7 GB maximum)
May  5 13:05:05 kernel:               0xffffffbe00000000 - 0xffffffbe00e00000   (    14 MB actual)
May  5 13:05:05 kernel:     fixed   : 0xffffffbffabfd000 - 0xffffffbffac00000   (    12 KB)
May  5 13:05:05 kernel:     PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
May  5 13:05:05 kernel:     modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
May  5 13:05:05 kernel:     memory  : 0xffffffc000000000 - 0xffffffc040000000   (  1024 MB)
May  5 13:05:05 kernel:       .init : 0xffffffc0006cd000 - 0xffffffc000706000   (   228 KB)
May  5 13:05:05 kernel:       .text : 0xffffffc000080000 - 0xffffffc0006cc204   (  6449 KB)
May  5 13:05:05 kernel:       .data : 0xffffffc000707000 - 0xffffffc0007d2180   (   813 KB)
May  5 13:05:05 kernel: Preemptible hierarchical RCU implementation.
May  5 13:05:05 kernel: NR_IRQS:64 nr_irqs:64 0
May  5 13:05:05 kernel: init_timers: timer_function_timeout virt addr 0xffffffc000721308
May  5 13:05:05 kernel: Architected cp15 timer(s) running at 50.00MHz (phys).
May  5 13:05:05 kernel: clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
May  5 13:05:05 kernel: sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
May  5 13:05:05 kernel: BRCM Legacy Drivers' Helper, all legacy drivers' IO memories/interrupts should be remapped here
May  5 13:05:05 kernel:      Remapping interrupts...
May  5 13:05:05 kernel:              hwirq      virq
May  5 13:05:05 kernel:                61         5
May  5 13:05:05 kernel:                64         6
May  5 13:05:05 kernel:                66         7
May  5 13:05:05 kernel:                91         8
May  5 13:05:05 kernel:                92         9
May  5 13:05:05 kernel:                93        10
May  5 13:05:05 kernel:                77        11
May  5 13:05:05 kernel:                78        12
May  5 13:05:05 kernel:                79        13
May  5 13:05:05 kernel:                80        14
May  5 13:05:05 kernel:                81        15
May  5 13:05:05 kernel:                82        16
May  5 13:05:05 kernel:                83        17
May  5 13:05:05 kernel:                84        18
May  5 13:05:05 kernel:                85        19
May  5 13:05:05 kernel:                86        20
May  5 13:05:05 kernel:                88        21
May  5 13:05:05 kernel:                96        22
May  5 13:05:05 kernel:                97        23
May  5 13:05:05 kernel:                98        24
May  5 13:05:05 kernel:                99        25
May  5 13:05:05 kernel:               118        26
May  5 13:05:05 kernel:               119        27
May  5 13:05:05 kernel:               106        28
May  5 13:05:05 kernel:               104        29
May  5 13:05:05 kernel:               105        30
May  5 13:05:05 kernel:                76        31
May  5 13:05:05 kernel:               120        32
May  5 13:05:05 kernel:               110        33
May  5 13:05:05 kernel:               148        34
May  5 13:05:05 kernel:               154        35
May  5 13:05:05 kernel:               155        36
May  5 13:05:05 kernel:               156        37
May  5 13:05:05 kernel:               157        38
May  5 13:05:05 kernel:               158        39
[... 890 lines cut ...]
Jun 21 13:48:12 hostapd: eth6: STA 18:31:bf:47:e1:39 RADIUS: starting accounting session 5335CB545A2A330C
Jun 21 13:48:12 hostapd: eth6: STA 18:31:bf:47:e1:39 WPA: pairwise key handshake completed (RSN)
Jun 21 13:48:12 kernel: wfd_registerdevice Successfully registered dev wds0.0.4 ifidx 1 wfd_idx 0
Jun 21 13:48:12 kernel: Register interface [wds0.0.4]  MAC: 04:d4:c4:50:4f:e8
Jun 21 13:48:12 kernel: device wds0.0.4 entered promiscuous mode
Jun 21 13:48:12 kernel: br0: port 9(wds0.0.4) entered listening state
Jun 21 13:48:12 kernel: br0: port 9(wds0.0.4) entered listening state
[... cut ...]
Jun 21 13:48:19 kernel: CFG80211-ERROR) wl_cfg80211_change_station : WLC_SCB_AUTHORIZE sta_flags_mask not set 
Jun 21 13:48:19 hostapd: eth6: STA d0:d2:b0:d6:4b:2b RADIUS: starting accounting session 92FEB9A846D5D372
Jun 21 13:48:19 hostapd: eth6: STA d0:d2:b0:d6:4b:2b WPA: pairwise key handshake completed (RSN)
Jun 21 13:48:21 crond[1143]: time disparity of 1120363 minutes detected
Jun 21 13:48:49 kernel: bcm_i2c: bus 0: Failed to detect SFP: 100 retries exhausted
Jun 21 13:53:12 WLCEVENTD: Deauth_ind D0:D2:B0:D6:4B:2B
Jun 21 13:53:12 hostapd: eth6: STA d0:d2:b0:d6:4b:2b IEEE 802.11: disassociated

jepan, 2020-06-24:
Updates:
June 22- Made RT-AX88U in repeater mode 
June 23 - no reboot
June 24 - no reboot (uptime: 1 day 18hrs the longest uptime with no random reboots)

jepan, 2020-06-27:
June 26- After making RT-AX88U as main router and with aimesh node RT-AC68U. In less than 24hrs it rebooted on its own again.
June 27 - removed RT-AC68U aimesh node. Will observe if main router RT-AX88U will reboot.

jepan, 2020-06-30:
Conclusion.
So after removing aimesh node (RT-AC68U) main router RT-AX88U does not encounter random reboots. I am not really sure what causes the random reboots but I guessing it has something to do compatibility between main router and aimesh node. Hopefully it will be fixed in new firmware releases.

asaddon, 2020-07-01:
If it's due to AiMesh then you need to contact Asus directly because it's closed source and there's nothing Merlin can do about it.

jepan, 2020-07-01:
Yes. I will be closing this now. Since issue with main router is resolved.

closed 2020-07-01 by jepan