RMerl/asuswrt-merlin.ng #541: What happened to 3.0.0.4_386_37589-g0b46fd4 ?

diegocr, 2020-06-28:
Hello,

Just wondering, a month ago or so there was available version 386 from asus.com, i upgraded one of the routers (AiMesh node) with it and everything was fine at first glance, although shortly afterwards i started noticing a high CPU usage on it, which eventually ceased though.

The day after, then realized the router automatically did rollback to 384 (didn't knew such capability existed) and 386 vanished from the site.

Then, a week ago noticed version 386 was put back in the asus site, but attempting to download it yield a 404, and now this version is gone again.

So.. wondering what happened, and/or what's going on.. just got a little paranoid thinking the asus site may got hacked...

paldier, 2020-06-28:
386 is the version number of the zen series, which model do you use.

RMerl, 2020-06-28:
Ask Asus, this has nothing to do with Asuswrt-Merlin.

closed 2020-06-28 by RMerl

diegocr, 2020-06-28:
@paldier I don't have  Zen-series laptop.. the router is one of those RT-AC....

@RMerl Ok, sorry, but already did the same day and still got no reply, so was just looking here for help... btw, i thought this is the source code repo for the firmware Asus does ship these days :-/

RMerl, 2020-06-28:
No, this repository is for the Asuswrt-Merlin project, which is a fork of the official firmware.  Asuswrt is developed internally by Asus and is managed completely separately by them, and they are the ones managing their update servers as well.

diegocr, 2020-06-28:
I see, then please accept my apologies.

Could you please remove this ticket? :)