RMerl/asuswrt-merlin.ng #542: Asus RT-AC68U - 384.17 FW - /jffs/configs/igmpproxy.conf no loading

guimbitencourt, 2020-06-28:
**Router:** Asus RT-AC68U
**FW:** 384.17
**Problem:** Can't load igmpproxy.conf file from /jffs/configs
**Description:** Have reflashed the fw and wiped NVram, if I do activate Igmp Proxy on lan tab it works flawless, however, if I disable it and try load a file from /jffs/configs it provides the following error: 

**There must be at least 2 Vif's where one is upstream.**

For testing purpose I copied the content on /tmp/igmpproxy.conf auto generated by the router to the /jffs/configs file and the error persist.

RMerl, 2020-06-28:
The issue tracker is only for development purposes.  If you need technical support or have a question, please use the support forums at https://www.snbforums.com/forums/asuswrt-official.51/ instead.

closed 2020-06-28 by RMerl

guimbitencourt, 2020-06-28:
Hello RMerl, thank you for the answer.

I am sorry to ask but, isn't it a bug in the igmpproxy version on the router fw? Just asking because as the file is an exactly copy of the auto generated file from the router it should work.

Best regards,

RMerl, 2020-06-28:
If running it through the webui works and running it manually through your own config file doesn't, then something must be different in your config file.

Here, enabling it on my RT-AC66U_B1 generates the following config file.

```
phyint eth0 upstream ratelimit 0 threshold 1 altnet 0.0.0.0/0
phyint br0 downstream ratelimit 0 threshold 1
```

I just stopped it, created a copy of that config, and manually launched it - works normally for me.

So, make sure you do have both interfaces defined, as specified in your error message.

guimbitencourt, 2020-06-29:
RMerl, 

Thanks again for your help, just flashed both Asus and yours 384.18 beta and both worked, downloaded again 384.17 and it worked flawless, probably a corrupt update from my side.

Sorry to bother you and thanks a lot for the excellent job, running 384.18 final already.

Best regards,