RMerl/asuswrt-merlin.ng #543: UI unsable after upgrading to 384.18 on RT-AX88U

iditude, 2020-06-29:
UI is not showing controls and left panel (see screenshot). I suspect a CSS issue.

I had to use recovery mode, flash back with Asus Stock firmware to recover controls. I tried flashing 384.18 again and same result. Previous firmware worked perfectly

Thanks

[image]

RMerl, 2020-06-29:
This is usually caused by broken Javascript.  Open your browser console and look for any Javascript error.

Also try changing language to English - sometimes Asus messes up the dict files by inserting unescaped quotes in some of the translations.

jay-79, 2020-06-30:
Hi,
I was facing the same issue. It's indeed a corrupt FR dict file's that cause this issue. Some unescaped quotes broke the rendered UI asp file (e.g. "S'incrire" on line 457 in index.asp), make Javascript broken and stop its execution with "undefined flag variable" error messages.

As a workaround, I switched to English language by SSH connect to my router and use the following commands :

```shell
nvram set preferred_lang=EN
nvram commit
```

Just refresh browser page afterward and it work like a charm.

iditude, 2020-06-30:
Hello,

Thanks it was indeed the language. I couldn't connect through SSH as it was activated before and the new UI didn't allow me to make any change so I had to do the whole recover / upgrade piece, but I'm back up now. 

As this seems like an ASUS issue, I think we can close? Or is there anything to do here?

Cheers

RMerl, 2020-06-30:
As it's a fairly disruptive issue I will look into fixing it on my end.

irixmo, 2020-06-30:
I've also seen exactly this on 384.17 today (AC68U / Chrome 83.0.4103.116), so not necessarily 384.18 related. Digging around the console errors it seems unable to load basic.css:

[image]

This is on my main unit, in Wireless Router mode. A second AC68U (also 384.17) in AP mode doesn't have the issue.

I've recently also been suffering from another UI issue (https://www.snbforums.com/threads/cant-login-to-gui.64631/) in case it has any bearing on this problem - couldn't access webui or ssh. Reboot solved that and that issue hasn't come back yet, but now seeing this problem instead.

RMerl, 2020-07-01:
Fixed with 2285ebbcc0d83a706417013dbf8026c9e6d869af .

closed 2020-07-01 by RMerl