RMerl/asuswrt-merlin.ng #547: * opkg_download: Failed to download http://bin.entware.net/armv7sf-k2.6/Packages.gz, wget returned 8.

tkalfaoglu, 2020-07-01:
I am unable to download anything..
# opkg update
Downloading http://bin.entware.net/armv7sf-k2.6/Packages.gz
*** Failed to download the package list from http://bin.entware.net/armv7sf-k2.6/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/armv7sf-k2.6/Packages.gz, wget returned 8.

My IP is in the range of  78.186.110.x   -- static IP

kissglay, 2020-07-01:
This is not a FW issue. It's simply because the site is down for the moment.

jackyaz, 2020-07-05:
Please do not use Github's Issue tracker for support requests. Use the support forums at [SNBForums](https://www.snbforums.com/forums/asuswrt-merlin.42/) instead. The issue tracker is for development/bug tracking purposes only!

closed 2021-01-03 by RMerl