RMerl/asuswrt-merlin.ng #557: Version 384.18 : issue in administration/system page when admin user already contained an underscore

QuantumMaster, 2020-07-07:
running Asus RT-AC88U with version 384.18.

Seems like in 384.18, if you want to update router parameters, but your admin user contained an underscore in its name, you cannot apply settings anymore. (i created my admin account already 1 year ago when i bought my AC88U)
"Only alphanumeric characters and dash symbol are accepted. The first character cannot be a dash - ."

JackMerlin, 2020-12-30:
The simple solution is to reset your router and set it up again!

hurda, 2020-12-30:
That's not a solution.
A solution would be ASUS adding a way to the firmware to force the user to change the username after booting the stricter firmware-version for the first time, or show a proper warning upon booting to do a full reset of the settings.
Doing nothing like that and just showing a rudimentary and unhelpful error-message is rather user-hostile.

JackMerlin, 2020-12-30:
>  A solution would be ASUS adding a way to the firmware to force the user to change the username after booting the stricter firmware-version for the first time, or show a proper warning upon booting to do a full reset of the settings.

They did this. If you reset your router after you log in for the first time, your router will let you set a new username and password, and your router will warn you about unavailable characters.

If you need more help, please ask in the SNBforums. 

Note: I am only a user, not a developer, not that merlin

hurda, 2020-12-31:
>They did this. 

No.

>If you reset your router after you log in for the first time

And why should a user do this without being notified of that change to username-validation?
The changelog of the ASUS-firmware only mentioned "Fixed login bugs."
No warning that the username might have been rendered invalid with the update.

JackMerlin, 2021-01-01:
If you think this is an Asus problem, please email them. It is useless to ask a question here. No technical support is provided here, please go to the SNB forums for technical support.

QuantumMaster, 2021-01-01:
Thank you for your comments.
Not sure how an end-user can distinguish if an error/bug is to be linked to the original Asus-Firmware and errors introduced by the Merlin mod's? (and, as such know where (Asus <-> Merlin) to open a bug issue)

FYI: In the end, a reboot is actually not needed, just changing your admin user-id in the system administration screen  to be compliant with the new 'allowed characters' does the trick.

Thanks again for the interactions and feed back!

closed 2021-01-01 by QuantumMaster