RMerl/asuswrt-merlin.ng #562: Bell FTTH doesnt reconnect when link drop

jonathanmasseurca, 2020-07-16:
I installed a RT-AX88U with MC220L and removed the HH3000.

Everything works fine except sometime the PPPoE link drop after a random time.

Sometime it work for few days and then it drop off. (On the router interface it say Disconnected)

I need to reboot the router everytime that it happen, Is there something that I can enable to auto-reconnect? I'm using Asuswrt-Merlin version 384.18.

When it happen I have those event :

Jul 8 02:00:41 pppd[510]: Connection terminated.
Jul 8 02:00:41 pppd[510]: Modem hangup

I also saw on this forum that people with asus-wrt meet the same issue.

https://www.snbforums.com/threads/script-to-reconnect-wlan-if-dropped.20313/

It would be nice to have a checkbox or something that detect that the pptp link has been dropped and make a reconnect without having to reboot.

Syst3mSh0ck, 2020-07-16:
Does this happen with the official firmware? It would be helpful to see the IPCP log from pppd but I'm unsure how to do this with AsusWRT. Maybe someone else can assist.

paldier, 2020-07-17:
Enable Internet Detection, PPP echo

jonathanmasseurca, 2020-07-17:
I didnt try the original firmware, I will try it,

My Pppoe settings are 
[image]

paldier, 2020-07-18:
99999? so bad, 10 or 15 is better, which means 60 or 90 seconds

jonathanmasseurca, 2020-07-18:
Thanks, i made the change, maybe this it, i will test it

closed 2020-07-18 by jonathanmasseurca