RMerl/asuswrt-merlin.ng #563: [DDNS Service]custom server

tobywu6652, 2020-07-18:
1. In the DDNS tag, if I set **Server** as **custom**, it'll say,
   > For the Custom DDNS you must manually create a ddns-start script that handles your custom notification.

   I think it will be better if there is a configuration wizard or upload button below.

2. If I haven't add any  ddns-start scripts, and set **Enable the DDNS Client** as **no**, it'll still say, "Applying Settings..."
   Does this need to be optimized？

lf9421, 2020-11-23:
在DDNS标记中，如果我将Server设置为custom，则会说，

对于自定义DDNS，您必须手动创建用于处理自定义通知的ddns-start脚本。
我也是同样的情况

JackMerlin, 2020-12-30:
>     1. I think it will be better if there is a configuration wizard or upload button below.

Use WinSCP or any software you like to upload to JFFS.

>     2. And it will never stop.

Just click the refresh of your browser.

Please refer to wiki for DDNS script:
https://github.com/RMerl/asuswrt-merlin.ng/wiki/User-scripts
https://github.com/RMerl/asuswrt-merlin.ng/wiki/Custom-DDNS
https://github.com/RMerl/asuswrt-merlin.ng/wiki/DDNS-services