RMerl/asuswrt-merlin.ng #565: host name is already in use for macOS devices

rdev02, 2020-07-22:
I've migrated to AC68U_384.17 a month ago from the stock firmware. One issue that came up with all of my macOS devices is that I keep hitting "[Computer name already in use](https://discussions.apple.com/thread/5018253)". The counter is in the hundreds now.
I'm using Guest and 2.4 and 5Hz wifi, along with YazFi

crkinard, 2020-07-22:
I remember this being more of an annoying MacOS thing than anything to do with any routing hardware.

rdev02, 2020-07-22:
might well be. The onset and the spread of the issue(we have about 5 different model and slightly different patch version) macOSs matches very well. I'm sure this is some edge case, which is not handled on one of the sides. Might be a YazFi issue, but YazFi doesn't have an issue tracker attached. Will ask on SNB.
Today one of my IoT devices stopped being able to access internet and I can't really figure out if it's the same issue, just the device isn't capable of handling this graciously.
I guess one way to verify is to go back to stock and see if things improve. Might do that over the weekend.

BenSokol, 2020-08-04:
I am running an RT AC88U with merlin 384.18. I am not sure when it started (which version), but i know it has been occuring since at least 384.16. I can also confirm that all apple devices are effected (Apple TV, iPhone, mac, even Apple Watch). This does not happen on an old Netgear router i tried, but i have not tried stock asus firmware.

Current setup:
amtm: v3.1.7
Diversion v4.1.12 (NOTE: I know this is unrelated as i only set it up very recently)
scribe: v2.4.3
connmon: v2.6.0
scMerlin: v1.1.2
spdMerlin: v3.6.1
uiDivStats: v2.2.1 (Unrelated, see Diversion)
uiScribe: v1.3.1

NOTE: I do NOT have YazFi setup, however i do have a couple guest networks. My mac only connects to the main network (WiFi) and never over Ethernet. I have both the 2.4G and 5G networks with the same name as well.

rdev02, 2020-08-04:
few days ago I have uninstalled YazFi and haven't been seeing those issues since... I did update the FW as well :(. Will wait for a bit more to see if it comes again.

BenSokol, 2020-09-22:
I am still seeing this. I wonder if it is related to having a custom domain name for network (i.e. local.example.com)

JackMerlin, 2020-12-30:
> but YazFi doesn't have an issue tracker attached

Each addons is maintained by a different developer, and you won’t get any help asking questions here. Please report the ddons problem in the [SNB forum](https://www.snbforums.com/forums/asuswrt-merlin-addons.60/).