RMerl/asuswrt-merlin.ng #566: DNS-over-TLS fails sometimes

n76, 2020-07-24:
Firmware: 384.18
Hardware: RT-AC1900P
Configured DNS-over-IP servers: 
2606:4700:4700::1111    cloudflare-dns.com
1.1.1.1    cloudflare-dns.com
1.0.0.1    cloudflare-dns.com
2606:4700:4700::1001    cloudflare-dns.com

DNS servers picked using "Preset servers" drop down.

On a computer within the LAN try a lookup for a DNS name that returns a large number of A records. Results vary, but for Linux based I sometimes get this:
```
$ nslookup archive.raspberrypi.org
;; connection timed out; no servers could be reached
```
and other times I get this:
```
$ nslookup archive.raspberrypi.org
;; Truncated, retrying in TCP mode.
;; communications error to 10.7.52.1#53: connection reset
```
On a MacOS I get:
```
$ nslookup archive.raspberrypi.org
;; Truncated, retrying in TCP mode.
;; communications error to xxxx:xxxx:4000:30::1#53: connection reset
```

With DNS-over-TLS I get successful response on both platforms with quite a few entries:
```
$ nslookup archive.raspberrypi.org | grep Address | wc -l
25
```
I suspect a buffer size issue someplace.

GitStorageOne, 2020-08-10:
I use DNS addresses from different dns providers:  
1.1.1.1  
8.8.8.8  

When i go to page [https://1.1.1.1/help](https://1.1.1.1/help), router reports warnings (two lines):  

> Insecure DS reply received for cloudflare.net, check domain configuration and upstream DNS server DNSSEC support
> Insecure DS reply received for cloudflare.net, check domain configuration and upstream DNS server DNSSEC support

I saw, that tester above may not work properly, but, i see as well
`AS Name` = `Google`, which, probably, mean that second DNS address is used as fall back.  
If so, then lookup performance is reduced.  

If only first DNS address will be entered, then tester will show `AS Name` = `CloudFlare` as expected.

n76, 2020-08-25:
The release notes for 384.19 (14-Aug-2020) include the following:
`  - FIXED: Some large DNS queries could fail when using DoT
           (patch backported from upstream)`

And since upgrading to this version I have not seen my problem reoccur.

Should this issue be closed?

RMerl, 2020-08-25:
Yes, since this issue is specifically about those longer replies.  The upstream fix isn't perfect (it relies on compression rather than address the core issue), but for now it's "good enough" until upstream gets a better fix in place.

closed 2020-08-25 by RMerl