RMerl/asuswrt-merlin.ng #578: 160MHz (5GHz) Bandwidth on RT-AC88U

Germandoe, 2020-08-09:
Since a few recent updates, my RT-AC88U has been showing 160MHz option under Wireless. I've tested it and the 5GHz wifi does not intialize on the 160MHz setting, It works flawlessly on 80MHz or lower tho.

Could this be an Asus bug? I have the latest Beta version at the moment, however this issue was present on 384.18 as well.

[image]

Have a good day and thank you for your software!

paldier, 2020-08-13:
you should use a non-china version device, these wifi chips are different models, but they are all named 4366 or 4366e

npmstart-pray, 2020-08-13:
You know the old story about the person who goes to see their doctor with the complaint “it hurts when I do this”?  The doctor tells the patient “well don’t do that”. 
You should consider following the doctor’s advice. Or, check the original specification of the router to determine if it will indeed support 5ghz wifi with 160MHz.

Germandoe, 2020-08-14:
> you should use a non-china version device, these wifi chips are different models, but they are all named 4366 or 4366e

The one I have is the EU version. Odd.

> You know the old story about the person who goes to see their doctor with the complaint “it hurts when I do this”? The doctor tells the patient “well don’t do that”.
> You should consider following [...]

Sounds about right. Tho I secretly wish ASUS added 160MHz support 😄 The RT-AC88U has only 80MHz out-of-the-box.

paldier, 2020-08-24:
> The one I have is the EU version. Odd.
>
>
> Sounds about right. Tho I secretly wish ASUS added 160MHz support 😄 The RT-AC88U has only 80MHz out-of-the-box.

The EU version hardware support 160, but the number of channels is not enough, you have to change the territory_code to AA/US/AU in cfe, i don't think you like this.

RMerl, 2020-09-21:
Known issue probably caused by the frankenbuild required to support that model in 384.19, I'd expect it to be fixed in future releases.

closed 2020-09-21 by RMerl