RMerl/asuswrt-merlin.ng #580: IPv6 CleanBrowsing 1/2 (family) DoT preset IP not recognized as valid

robertdonat, 2020-08-16:
After choosing the preset IPv6 addresses of CleanBrowsing 1 and 2 (family) and clicking add, error message of "2a0d:2a00:X:: is not a valid IP address!" appears. I confirmed these are the correct [IPv6 addresses of CleanBrowsing Family](https://cleanbrowsing.org/guides/dnsovertls). 

- IPv6 CleanBrowsing 1: 2a0d:2a00:1:: - family-filter-dns.cleanbrowsing.org
- IPv6 CleanBrowsing 2: 2a0d:2a00:2:: - family-filter-dns.cleanbrowsing.org

The other preset IPv6 address for CleanBrowsing are accepted.

[image]
[image]

paldier, 2020-08-16:
>     2a0d:2a00:1::
It is not a valid IP address, what is your question, firmware bug? NO!

robertdonat, 2020-08-16:
> It is not a valid IP address, what is your question, firmware bug? NO!

2a0d:2a00:1:: is a valid IPv6 address. Check your work before commenting, please.

[image]

closed 2020-10-12 by RMerl

hurda, 2020-10-12:
Isn't there a bug in the IPv6-validator?
https://github.com/RMerl/asuswrt-merlin.ng/blob/5f61bfdfd4305288a6c2258a8d34d41be8624f25/release/src/router/www/validator.js#L846
It splits the address into a part left of :: and a part right of :: .
Then it checks the strings separated by : in these parts if they are only made of at least 1 or at most 4 hexadecimal characters with the isHex-function.
But if there is no part right of :: , the check fails, although such an address is valid.
Am I reading this correctly?

commit a5d6583c (2020-10-12): rom: include trailing 0 to the IPv6 addresses of Cleanbrowsing Family (closes #580)