RMerl/asuswrt-merlin.ng #582: AiCloud page shows wrong hostname when used with Namecheap DDNS

mohd-akram, 2020-08-17:
On the DDNS page, I'm using Namecheap with the following:
```
  Host Name: my.router
Domain Name: example.com
```
On the Network Map page, it correctly shows `DDNS: my.router.example.com`. However, on the AiCloud page it says: `Enter AiCloud 2.0 https://my.router`. I also can't access AiCloud in the first place, I'm not sure if it's because of this or something else.

mohd-akram, 2020-08-17:
This does in fact seem to be the problem with not being able to access AiCloud too. Faking the hostname with `curl --resolve my.router:my.router.example.com https://my.router` succeeds.

gitstua, 2020-09-20:
Did you check the port - I think you need to specify in this format: https:example.com:8443

JackMerlin, 2020-12-30:
I guess it is a port problem, HTTPS uses the default port 8443, AiCloud uses the default port 443, BTW, AiCloud issues please do not feedback here, because it is a closed source component of Asus, RMerlin can't do anything.

mohd-akram, 2020-12-30:
It's not a AiCloud/port problem or else the curl command I used wouldn't work. It's a problem with the Namecheap DDNS.

mohd-akram, 2021-03-17:
After updating to 386.1, this now works.

closed 2021-03-17 by mohd-akram