RMerl/asuswrt-merlin.ng #583: Enabled "Adaptive QoS" generate error "ISP's DHCP did not function properly"

askuzminov, 2020-08-18:
Recently, after the end of DHCP Lease time, my WAN (Automatic IP) with Adaptive QoS enabled does not work for me. Firmware versions: 384.19, 384.18 and older.
Router: RT-AC68U.

I reset to factory settings, changed WAN parameters, DHCP query frequency. But nothing helped.

If I turn off Adaptive QoS everything works.

Here is the log:
```
Aug 16 16:07:56 lldpd[315]: removal request for address of XXX.XXX.XXX.XXX%4, but no knowledge of it
Aug 16 16:07:56 miniupnpd[6352]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Aug 16 16:07:56 dnsmasq[244]: read /etc/hosts - 7 addresses
Aug 16 16:07:56 miniupnpd[6352]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Aug 16 16:07:56 miniupnpd[6352]: Failed to get IP for interface eth0
Aug 16 16:07:56 miniupnpd[6352]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Aug 16 16:07:56 wsdd2[6349]: error: wsdd-mcast-v4: wsd_send_soap_msg: send
Aug 16 16:07:56 dnsmasq[244]: read /etc/hosts.dnsmasq - 11 addresses
Aug 16 16:07:56 dnsmasq[244]: using nameserver 127.0.1.1#53
Aug 16 16:07:56 dnsmasq[244]: using nameserver XXX.XXX.XXX.XXX#53 for domain beeline 
Aug 16 16:07:56 dnsmasq[244]: using nameserver XXX.XXX.XXX.XXX#53 for domain beeline 
Aug 16 16:08:00 WAN_Connection: ISP's DHCP did not function properly.
```

I searched for the cause of this error for a long time and found it in iptables.
When you enable Adaptive QoS, these rules are added.

```
/* DPI engine */
if (IS_AQOS()) {
    eval("iptables", "-t", "mangle", "-N", "BWDPI_FILTER");
    eval("iptables", "-t", "mangle", "-F", "BWDPI_FILTER");
    eval("iptables", "-t", "mangle", "-A", "BWDPI_FILTER", "-i", wan_if, "-p", "udp", "--sport", "68", "--dport", "67", "-j", "DROP");
    eval("iptables", "-t", "mangle", "-A", "BWDPI_FILTER", "-i", wan_if, "-p", "udp", "--sport", "67", "--dport", "68", "-j", "DROP");
    eval("iptables", "-t", "mangle", "-A", "PREROUTING", "-i", wan_if, "-p", "udp", "-j", "BWDPI_FILTER");
}
```
source code: https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/rc/firewall.c#L5725

They block the WAN.

I added a custom script **/jffs/scripts/firewall-start** to disable these rules:

```
#!/bin/sh
if [ "$(nvram get qos_enable)" = "1" ] && [ "$(nvram get qos_type)" = "1" ]; then
  iptables -t mangle -D BWDPI_FILTER 1
  iptables -t mangle -D BWDPI_FILTER 1  
  iptables -t mangle -I BWDPI_FILTER 1 -i eth0 -p udp -m udp --sport 67 --dport 68 -j ACCEPT
fi
```

And now everything works correctly.

Why are these rules needed? Why are they needed only for Adaptive QoS, but not for Traditional QoS or with QoS disabled?

dave14305, 2020-09-09:
It’s discussed in the wiki and on SNBforums.com. It’s Asus’ design, unknown reasons.

https://github.com/RMerl/asuswrt-merlin.ng/wiki/Adaptive-QoS-Optimization#known-issues-with-adaptive-qos

askuzminov, 2020-10-09:
@dave14305 Maybe then it's better to just delete this code? Or add the fix directly to the RT-AC68U firmware.

RMerl, 2020-10-09:
I'm sure they had a valid reason for these.  I asked them what it is, we'll see.

For the record, I have zero problem with my own ISP and Adaptive QoS.  My lease gets renewed every day without failure.

RMerl, 2020-10-10:
I got Asus's answer, and basically they do it because Trend Micro does it in their reference documentation.  So, they don't know why it's there either.

RMerl, 2020-10-11:
```
admin@stargate88ax:/tmp/home/root# iptables -t mangle -L BWDPI_FILTER -vn
Chain BWDPI_FILTER (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 DROP       udp  --  eth0   *       0.0.0.0/0            0.0.0.0/0            udp spt:68 dpt:67
   27  9560 DROP       udp  --  eth0   *       0.0.0.0/0            0.0.0.0/0            udp spt:67 dpt:68
```

My connection is still working fine here.

I'm willing to try disabling it, we'll see the end-user feedback about this change once that code reaches public beta.  The fact that it still works for most people is puzzling.

RMerl, 2020-10-11:
Update: removing this won't be possible.  It also gets added by the closed source libbwdpi library...

RMerl, 2020-10-12:
I`ve pointed an Asus engineer at this discussion.  He'll review things on his end, I will let you know if I hear anything else.

RMerl, 2020-10-12:
@askuzminov can you send me an email (merlin <at> asuswrt-merlin <dot> net)?  I will put you in contact with the Asus engineer who works on the QoS code for further troubleshooting (he'll probably need you to do some packet capturing).

@dave14305  are you able to reproduce the problem?  If so, contact me as well.

dave14305, 2020-10-12:
> @askuzminov can you send me an email (merlin asuswrt-merlin net)? I will put you in contact with the Asus engineer who works on the QoS code for further troubleshooting (he'll probably need you to do some packet capturing).
> 
> @dave14305 are you able to reproduce the problem? If so, contact me as well.

I cannot, but @solstyce9 experienced the problem and wrote the wiki article linked earlier.

RMerl, 2020-10-12:
@solstyce9 can you also email me please if you are still able to reproduce this issue, and want to help track it down?  Thanks.

dave14305, 2020-10-12:
> @solstyce9 can you also email me please if you are still able to reproduce this issue, and want to help track it down? Thanks.

This post has some iptables logging of his issue back in May:
https://www.snbforums.com/threads/release-freshjr-adaptive-qos-improvements-custom-rules-and-inner-workings.36836/page-214#post-581023

RMerl, 2020-10-12:
Thanks.  I suspect tcpdump captures will be more useful to the Asus engineer thought, so I'd prefer to put the two of them in touch.  tcpdump could be installed through Entware.

solstyce9, 2020-10-13:
Sorry, when I updated my service, Sonic gave me a new gateway that doesn't
have the same problem so I can't reproduce it any longer.  It was
definitely an issue with the specific gateway though.

askuzminov, 2020-10-13:
Additional Information:

My ISP (Russia, Beeline) has a very short Lease time = 10 minutes.

0 - 10 minutes - The first connection is working correctly.
10 - 20 minutes - Communication is lost with "ISP's DHCP did not function properly" error.
20 - 30 minutes - The connection appears, the Internet works.
30 - 40 minutes - Communication is cut off again.
...
And so it alternates.

RMerl, 2020-10-13:
Thanks, I just emailed both you and the Asus engineer who will be investigating with you.

RMerl, 2020-10-13:
> Sorry, when I updated my service, Sonic gave me a new gateway that doesn't have the same problem so I can't reproduce it any longer. It was definitely an issue with the specific gateway though.

Gotcha, thanks for answering.

RMerl, 2020-10-20:
Asus workaround being tested.

RMerl, 2020-10-27:
Workaround working for one tester.

Asus already has a proper fix implemented upstream, but for now this workaround should resolve the issue until the fixed libbwdpi can be merged.

closed 2020-10-27 by RMerl