RMerl/asuswrt-merlin.ng #590: GoDaddy ddns-start wiki example doesn't work for domains with multiple A records

gxfgbo, 2020-08-25:
The ddns-start example on the [wiki](https://github.com/RMerl/asuswrt-merlin.ng/wiki/DDNS-Sample-Scripts#godaddy) replaces all A records for a domain. This will work for domains with only a single A record. Instead recommend the following script to push A record updates for domains with multiple A records:

#!/bin/sh

IP=${1}
APIKEY="INSERT_YOUR_GODADDY_API_APP_KEY_HERE"
SECRET_KEY="INSERT_YOUR_GODADDY_API_SECRET_KEY_HERE"
DOMAIN="INSERT_YOUR_GODADDY_DOMAIN_NAME_HERE"
HOST="INSERT_HOSTNAME_TO_UPDATE_HERE"

curl -X PUT https://api.godaddy.com/v1/domains/$DOMAIN/records/A/$HOST \
  -H "Authorization: sso-key $APIKEY:$SECRET_KEY" \
  -H "Content-Type: application/json" \
  -d "[{\"type\": \"A\",\"name\": \"$HOST\", \"ttl\": 600, \"data\": \"$IP\"}]"

if [ $? -eq 0 ];
then
    /sbin/ddns_custom_updated 1
else
    /sbin/ddns_custom_updated 0
fi

RMerl, 2020-08-25:
Go ahead and edit it, everyone has write access to the Wiki.

gxfgbo, 2020-08-26:
Done. Thanks!

closed 2020-09-21 by RMerl