RMerl/asuswrt-merlin.ng #594: is RT-AC87U model really deprecated ? Updates are coming from asus

nobless, 2020-08-30:
Version 3.0.0.4.382.52288
2020/07/0938.93 MBytes
ASUS RT-AC87U Firmware version 3.0.0.4.382.52288
Security update
- Fixed CVE-2020-12695 (CallStranger)
- Fixed Reflected XSS vulnerability.
- Fixed Directory traversal vulnerability.
- Fixed CVE-2017-15653.

asaddon, 2020-08-31:
These updates are for 382 codebase, Merlin moved to 384 codebase a long time ago and currently in process of moving again to 386 codebase. These codebases are not compatible with each other without the help of Asus providing the necessary binary blobs for the closed source components and they're not doing it anymore which leaves Merlin no choice but to drop support for such models. 

Also, use the Forum for such issues, the issue tracker is for development purposes only.

breisig, 2020-08-31:
I think still supporting RT-AC87U (even in very limited support) should still be done. Al least create one more firmware update for it that addresses all the issues above.

npmstart-pray, 2020-08-31:
> I think still supporting RT-AC87U (even in very limited support) should still be done. Al least create one more firmware update for it that addresses all the issues above.

The team working on Merlin is very small, and with only so many hours in a day to do unpaid work, some tough decisions must be made that you may not agree with. If you think that current firmware from Asus is better/whatever than Merlin, why not switch back?  If you’re concerned with losing functionality of some of the scripts you may be running, there are certain options for you to explore that may help you make a decision regarding a potential reversion to “move forward”. 
I was disappointed when my n66 moved to unsupported, but I was more disappointed when it ceased to function shortly after that. Some things happen for a reason, right?

RMerl, 2020-08-31:
> I think still supporting RT-AC87U (even in very limited support) should still be done. Al least create one more firmware update for it that addresses all the issues above.

I've already gone through a lot of efforts just to keep it supported through three more updates than was initially possible, so it already was on extended life support for quite a long time.  Supporting it is simply no longer possible nor realistic.  The Asus RT-AC87U code is no longer compatible with mine, and last attempt at making it work through hacks and patches failed, with the firmware crashing on boot.

RMerl, 2020-08-31:
And BTW, at least half of the fixes you mention don't even apply to my firmware.  CVE-2020-12695 for instance was never a problem with my firmware, as the miniupnpd daemon that I use it not vulnerable to it.

closed 2020-08-31 by RMerl