RMerl/asuswrt-merlin.ng #595: RT-AX88U connection speed

AlexanderPavlenko, 2020-08-30:
I recently noticed that download speed is way less than it used to be. Update to 384.19, factory reset – same issue. WAN is ISP router's LAN.

Ethernet via old Mikrotik connected to ISP router is all good:
Speedtest on client: 360 Mbps download, 360 Mbps upload

Ethernet via RT-AX88U:
Speedtest on client: 75 Mbps download, 360 Mbps upload
iperf -s on router, iperf -c on client: 946 Mbits/sec
iperf -c on router, iperf -s on client: 931 Mbits/sec

Wireless is affected too with the same asymmetry. No custom routes or VPN.
Please advice what could be the cause.

AlexanderPavlenko, 2020-08-30:
The issue seems to be mitigated by using Dual WAN feature and connecting both cables to ISP router.
Still not sure why RT-AX88U's main WAN port was limited.

chchia, 2020-09-20:
alright, so it wasn't just me then.

i have two ax router, the Asus RT-AX58u with merlin firmware download at about 400mbps, while the other none Asus router can download 500mbps.

BillFleming, 2020-10-04:
I don't appear to have this problem on mine. I just did a straight upgrade from 384.17 to 384.19. I don't have a super fast connection yet though (waiting on move). Still only Comcast business docsis around 115mbps currently.

davecarlson, 2020-11-07:
I too have noticed wifi speed drop on the AX88 with the most recent release. Cant get over 500mb on wireless, even close to the router with direct line of site, not other routers operating in the local air on the same channel. Ethernet speeds 960mb.

JackMerlin, 2020-12-30:
No offense, but for this kind of urban legendary problem, I suggest you report it on the SNBforums. In my experience, 90% is related to your settings rather than the developer's problem.

lorenc-tomasz, 2021-01-05:
> I too have noticed wifi speed drop on the AX88 with the most recent release. Cant get over 500mb on wireless, even close to the router with direct line of site, not other routers operating in the local air on the same channel. Ethernet speeds 960mb.

Similar visible drop on 386.x