RMerl/asuswrt-merlin.ng #603: Feature Request - support Trim SSD

rah66, 2020-09-09:
By default, the trim function does not work on firmware. There are two solutions:
1.add a device with the unmap option via the udev rule
https://wiki.archlinux.org/index.php/Solid_state_drive#External_SSD_with_TRIM_support
which is not possible due to the absence of this component
2.switch the scsi driver manually to unmap mode
https://www.snbforums.com/threads/solved-how-activate-trim-for-an-ssd-merlinwrt.66105/post-616137
the second way is completely functional

```
/jffs/scripts/[init-start]
#!/bin/sh
cru a TRIM_SSD "0 */4 * * * fstrim -a -v >> /tmp/syslog.log 2>&1" ; sleep 60 ; echo -n unmap > /sys/block/sda/device/scsi_disk/0:0:0:0/provisioning_mode
```
[image]

RMerl, 2020-09-09:
The busybox trim applet wasn't enabled, I only enabled it a few days ago.

rah66, 2020-09-09:
> The busybox trim applet wasn't enabled, I only enabled it a few days ago.

can you add udev?

RMerl, 2020-09-09:
No, I don't want to touch kernel options unless really necessary, to avoid compatibility issues with pre-compiled kernel modules.

The firmware is built around hotplug2, so no idea what kind of conflict this would create.

rah66, 2020-09-14:
> No,

Sorry for the importunity, but can you add the packages **hdparm** and **sg3_utils**? With the help of them, can check trim support and activate it, through the pre-mount. Or maybe you suggest your own version.

`PM="$(find /sys/block/${1:5:3}/device/ -name provisioning_mode 2> /dev/null)"

if [hdparm -I /dev/${1:5:3} | grep -qF 'Data Set Management TRIM supported'] && [sg_vpd -a /dev/${1:5:3} | grep -qF 'Unmap command supported (LBPU): 1']
   then
      echo -n unmap > $PM
fi`

RMerl, 2020-09-15:
These would be bloat in a router.  Install them through Entware if you need them.

rah66, 2020-09-16:
> Install them through Entware if you need them.

They are needed before loading the Entware....

kissglay, 2020-09-16:
> > Install them through Entware if you need them.
> 
> They are needed before loading the Entware....

I wonder are there any differences between enabling them in 'post-mount' and 'pre-mount'?

rah66, 2020-09-16:
> I wonder are there any differences between enabling them in 'post-mount' and 'pre-mount'?

There is no difference, I use the pre-mount because it works with the variable $1=/dev/sdXX, and I can find out exactly where the provisioning_mode is located and exactly use hdparm with sg_vpd

> Is the cron job still needed if you put it in 'pre-mount'?

No, it only works when connecting a new device. Cron is needed only for performing fstrim.

kissglay, 2020-09-16:
Would this serve your purpose if put in the 'post-mount'?
`PM=$(find /sys/block/$(mount | sed -rn "s/^\/dev\/(.{3}).+$1.+/\1/p")/device/ -name provisioning_mode 2> /dev/null)`

rah66, 2020-09-16:
> if put in the 'post-mount'?

Ok, its work for "unmap/provisioning_mode". How do i make sure the connected device supports TRIM without hdparm & sg_vpd?

kissglay, 2020-09-16:
Yes, it's only work with Entware...

rah66, 2020-09-16:
> Yes, it's only working with Entware...
> But this's sort of an addon feature.

I would like to pick out hdparm & sg3_utils and find out from where they choose the necessary parameters in order to do without them.

rah66, 2020-10-03:
After conducting a bunch of experiments, it was revealed:
1.The Trim works only with the EXT4 file system.
2. Hdparm cannot be filtering tools, it does not support, for example, NVME-USB bridge RTL9210
```
admin@RT-AX88U:/tmp/home/root# hdparm -I /dev/sdb

/dev/sdb:
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 00 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

ATA device, with non-removable media
Standards:
        Likely used: 1
Configuration:
        hard sectored
        soft sectored
        not MFM encoded
        spindle motor control option
        disk xfer rate > 5Mbs, <= 10Mbs
        disk xfer rate > 5Mbs
        rotational speed tol.
        track offset option
        Logical         max     current
        cylinders       11823   0
        heads           27746   0
        sectors/track   0       0
        --
        bytes/track: 25455      bytes/sector: 107
        Logical/Physical Sector size:           512 bytes
        device size with M = 1024*1024:           0 MBytes
        device size with M = 1000*1000:           0 MBytes
        cache/buffer size  = unknown
Capabilities:
        IORDY not likely
        Cannot perform double-word IO
        R/W multiple sector transfer: not supported
        DMA: not supported
        PIO: pio0

admin@RT-AX88U:/tmp/home/root# sg_vpd --page=0xb2 /dev/sdb
Logical block provisioning VPD page (SBC):
  Unmap command supported (LBPU): 1
  Write same (16) with unmap bit supported (LBPWS): 0
  Write same (10) with unmap bit supported (LBPWS10): 0
  Logical block provisioning read zeros (LBPRZ): 0
  Anchored LBAs supported (ANC_SUP): 0
  Threshold exponent: 0 [threshold sets not supported]
  Descriptor present (DP): 0
  Minimum percentage: 0 [not reported]
  Provisioning type: 0 (not known or fully provisioned)
  Threshold percentage: 0 [percentages not supported]
admin@RT-AX88U:/tmp/home/root#
```
However, it works fine and the Trim task is worked.
Normal output hdparm SATA-USB bridge JMS580:
```
admin@RT-AX88U:/tmp/home/root# hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
        Model Number:       CT240BX500SSD1
        Serial Number:      ************
        Firmware Revision:  M6CR013
        Media Serial Num:
        Media Manufacturer:
        Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
        Used: unknown (minor revision code 0x0110)
        Supported: 9 8 7 6 5
        Likely used: 9
[... 99 lines cut ...]
p.s. When connecting drives with the EXT4 file system and without Trim support, they work in the UNMAP mode, but the Trim is executed with an error according to the schedule. Not critical.
```
admin@RT-AX88U:/tmp/home/root# lsblk --discard
NAME       DISC-ALN DISC-GRAN DISC-MAX DISC-ZERO
sda               0        4K       4G         0
├─sda1            0        4K       4G         0
└─sda2            0        4K       4G         0
sdb               0      512B       4G         0
└─sdb1            0      512B       4G         0
mtdblock0         0        0B       0B         0
mtdblock1         0        0B       0B         0
mtdblock2         0        0B       0B         0
mtdblock3         0        0B       0B         0
mtdblock4         0        0B       0B         0
mtdblock5         0        0B       0B         0
mtdblock6         0        0B       0B         0
mtdblock7         0        0B       0B         0
mtdblock8         0        0B       0B         0
mtdblock9         0        0B       0B         0
mtdblock10        0        0B       0B         0
mtdblock11        0        0B       0B         0
admin@RT-AX88U:/tmp/home/root# fstrim -v /tmp/mnt/sdb1
fstrim: /tmp/mnt/sdb1: FITRIM ioctl failed: Input/output error
admin@RT-AX88U:/tmp/home/root#
```

closed 2020-12-07 by rah66