RMerl/asuswrt-merlin.ng #604: 4G-AC86U & 4G-AC68U support

purehamtortoisezebra, 2020-09-11:
Please consider adding support for [4G-AC86U](https://www.asus.com/Networking/4G-AC86U/) & [4G-AC68U](https://www.asus.com/Networking/4G-AC68U/). While these two models come with external antennas (>2), there's only built-in antenna on most external 4G LTE USB Modems which provides poor network performance. It's waste of ASUS's high-end hardware. In some countries/regions like Hong Kong, people use 4G LTE broadband widely because high-speed broadband can't be installed in some suburbs.

Adamm00, 2020-09-15:
https://www.snbforums.com/threads/faq-read-me-first-before-posting-a-question.47153/post-609193

purehamtortoisezebra, 2020-09-15:
However, these two models 4G-AC86U & 4G-AC68U corresponds to RT-AC86U & RT-AC68U which are supported by Asuswrt-Merlin currently. I think work to port Merlin to these models isn't complicated since they have same SOC as their corresponding models, just with 4G LTE module added.

RMerl, 2020-09-15:
They still require dedicated support for them, with their own GPL updates from Asus.  This firmware project requires closed source components that are unique to each single individual model, and therefore adding any more model is never simple.

closed 2020-09-15 by RMerl