RMerl/asuswrt-merlin.ng #609: Cannot connect to a particular website using 384.19

eraie, 2020-09-21:
AC1900 384.19 (AC68U fw): suddenly unable to connect to a particular site since upgrading to 384.19

No settings were changed during upgrade 384.18 > 384.19, very sudden issue. Unable to connect by IP either. Trace shows a timeout on an early hop connecting to this site. This renders the site inaccessible on all connected network devices. Tired various DNS, various MTU, no change. 

Despite the timeout, no other routers (or VPN running on same router), and no other external networks fail to connect to the site - unable to replicate the problem outside of 384.19

SNB thread:  [https://www.snbforums.com/threads/sudden-blocking-of-website-on-merlin-384-19.66412/#post-620084](url)

RMerl, 2020-09-21:
Nothing on the router firmware can block access to specific sites, unless you use a filtering DNS, or AiProtection's malicious website blocking.  You are more likely to be blocked by the remote server due to your current WAN IP, which might change between router reboots/flashes.

closed 2020-09-21 by RMerl

eraie, 2020-09-21:
Not at all using AiProtection, not at all using a filtering DNS, exact same WAN IP.  2 different routers with same WAN IP are connecting without issue...including older N66U on 380.70