RMerl/asuswrt-merlin.ng #612: IPv6 passthrough does not work when NAT hardware acceleration is on

reet-, 2020-09-26:
Hi,

As it seems IPv6 passthrough and NAT hw accl do not work together.

See also: 
https://www.snbforums.com/threads/ipv6-pass-through-doesnt-work-when-nat-hardware-acceleration-turns-on-ac68u.56267/
https://www.snbforums.com/threads/conflict-with-ipv6-passthrough-and-nat-hardware-acceleration.56522/

My firmware is 384.19.

On my RT-AC68U NAT Acceleration is set to Auto -> "CTF (Cut Through Forwarding) and FA (Flow Acceleration) accelerator are enabled".

Log is flooded with "kernel: osl_pktfastfree: free skb for NULL ctfpool" messages, connections are unstable. Disabling NAT acceleration is a bitter pill to swallow as I am on a Gbit uplink. Performance for IPv4 NAT drops from around 800/900mbit to around 200-300 mbit.

JackMerlin, 2020-12-30:
Try to run the official firmware, and report it to Asus if the same error occurs. CTF and any hardware acceleration are Broadcom's closed source code, and RMerlin can do nothing.

reet-, 2020-12-30:
Thanks for the information. I changed my setup from "Passthrough" to "Native", which was the correct setup for my ISP in the first place. This setup works with hwaccel now, closing the issue.

Cheers

closed 2020-12-30 by reet-