RMerl/asuswrt-merlin.ng #616: WireGuard support

purehamtortoisezebra, 2020-10-08:
Hello,

WireGuard is a much better VPN than OpenVPN. It's faster and more secure. And its ver1.0 has been released. Hope you consider adding WireGuard to Asuswrt-Merlin carefully. I would be very appreciated.

https://www.wireguard.com

crkinard, 2020-10-08:
Last I looked it only works on a small set of routers because of kernel version requirements.

npmstart-pray, 2020-10-08:
> Hello,
> 
> 
> 
> WireGuard is a much better VPN than OpenVPN. It's faster and more secure. And its ver1.0 has been released. In some countries like China, WireGuard works more stable than OpenVPN due to censorship. Hope you consider adding WireGuard to Asuswrt-Merlin carefully, at least popular RT-AX88U which comes with new Linux kernel supports WireGuard. I would be very appreciated.
> 
> 
> 
> https://www.wireguard.com

WireGuard works on AX88 and AC86. 
Look for the instructions on smallnetbuilder in the VPN forum. 
Merlin may never officially support it unless Asus chooses to.

purehamtortoisezebra, 2020-10-09:
Actually the ASUS router model I'm using is RT-AX88U. Its Linux kernel version should meet requirement of WireGuard. I have read threads about WireGuard installation and configuration by CLI on Smallnetbuilder. But what I need is ASUS Web GUI setting page of WireGuard just like OpenVPN. I understand many people ask for VPN protocols they personally prefer, but WireGuard will be used as widely as OpenVPN in the future. There're many VPN providers are offering/planning to offer WireGuard servers such as NordVPN, Mullvad VPN, AzireVPN, etc. Since I have no enough coding knowledge to contribute to Asuswrt-Merlin, hope you guys consider adding WireGuard to Web GUI carefully. Thanks.

purehamtortoisezebra, 2020-10-09:
The ASUS official prefers conservatism. They even don't add DNS-over-TLS to their firmware. I don't expect them to support WireGuard.

npmstart-pray, 2020-10-09:
If more people admit to using it, Asus might see the benefit to making it easy for their users. 
Sometimes it hurts to be way out on the bleeding edge, but somebody needs to do the work or we don’t move forward. Not you?  Why not?

Sent from nowhere near my ready room with my new tricorder

RMerl, 2020-10-09:
I don`t have plans in the short term for Wireguard support, due to the amount of work involved in re-implementing everything, and then in having to maintain it in addition to the OpenVPN support.

npmstart-pray, 2020-10-09:
Apologies - I was under the impression that WireGuard support was up to Asus to implement. 

Sent from nowhere near my ready room with my new tricorder

closed 2020-10-11 by purehamtortoisezebra

purehamtortoisezebra, 2020-10-11:
> WireGuard works on AX88 and AC86.
> Look for the instructions on smallnetbuilder in the VPN forum.
> Merlin may never officially support it unless Asus chooses to.

@RMerl I agree with @npmstart-pray that WireGuard support is ASUS's business because it's a huge job. Therefore it's better to ask ASUS to implement it. have sent email to ASUS to request WireGuard support. Thank you all☺️