RMerl/asuswrt-merlin.ng #617: dnsmasq[16108] error, websites not opening

Germandoe, 2020-10-09:
Hello,

I noticed a problem with my Galaxy S9+ and my RT-AC88U (on 384.19) on a daily basis. 
Often, websites don't load because of it. If I reconnect to WiFi, it loads just fine for the next 20min or so, then the error happens again. I tested this and 192.168.1.1 (and other local and external IPs) works just fine while the error is happening (indicating a DNS error). 
The only twist is that I have a DNS server set on my S9 as 1dot1dot1dot1.cloudflare-dns.com. If I disable this, the websites start working instantly.

The log is getting spammed with the following error while this happens:
_dnsmasq[16108]: failed to send packet: Operation not permitted_
_dnsmasq[16108]: failed to send packet: Operation not permitted_
_dnsmasq[16108]: failed to send packet: Operation not permitted_

**How to reproduce:**
1. Set 1dot1dot1dot1.cloudflare-dns.com as private DNS on phone, connect phone to router with the following DNS settings:
[image]
2. Wait some time, the error will magically appear. Direct IPs will continue to work (meaning DHCP is OK). Websites won't load. Disable the Android private DNS and boom, websites work again and log isn't getting spammed anymore. If you enable and disable WiFi for a brief second, the same result is achieved for some reason.

I do have "DNS-based Filtering" enabled though - but it isn't being used for the Galaxy S9, it's only being used for some IoT devices I own. Global Filter Mode is set to No Filtering.

Unsure how this can be connected. Please correct me if I'm wrong. But I was able to reproduce this countless times. Private Android DNS seems to always trigger it. No android DNS, no problem. But I kinda prefer using it because it adds an extra layer of security while outside my LAN. Also, no such error occurs while on 4G.

dave14305, 2020-10-10:
Better to post this configuration issue on snbforums.com. This isn’t a firmware development issue.

https://github.com/RMerl/asuswrt-merlin.ng#please-do-not-use-githubs-issue-tracker-for-support-requests--use-the-support-forums-at-snbforums-instead--the-issue-tracker-is-for-developmentbug-tracking-purposes-only

Germandoe, 2020-10-10:
> Better to post this configuration issue on snbforums.com. This isn’t a firmware development issue. [...]

I posted it here specifically because the router is blocking DNS resolves even tho the configuration is correct. Unsure but this seems like a bug to me.