RMerl/asuswrt-merlin.ng #618: Will this be supported for the new ASUS ZenWifi routers?

ksylvan, 2020-10-11:
Specifically:

- ZenWiFi AC (CT8)
- ZenWiFi AX (XT8)

Looking to bypass my uVerse BWG210-700 router setup with the new two node mesh setup I just bought.

Thanks!

RMerl, 2020-10-11:
I can't speculate about future model support, sorry.

closed 2020-10-11 by RMerl

paiou, 2020-11-29:
Hello.
I would be happy to contribute a merge request to support ZenWifi AX, I'm an experienced developer but never worked on driver or firmware code. Do you think this is achievable with reasonable effort?
Thanks!

bohtho, 2021-03-03:
Are you any closer to XT8 support? It’s a popular mesh system and I would love to get the same additional features on it. Is the missing support due to less “hackablility” and more “consumer black box”?

jmstacey, 2021-06-20:
Poking around this looks technically feasible as the XT8 has the same CPU and Radios 1+2 as the RT-AX56U, and radio 3 as GT-AX11000. Both of those routers have Merlin support so driver blobs are there. Don't know if there's any secret sauce for the Zen's compared to standalone router.