RMerl/asuswrt-merlin.ng #619: build error on WSL

MimoKar, 2020-10-11:
firmware build error :

from log : 

`touch pc-files
make[7]: Leaving directory '/home/mimo/amng.ax88/release/src/router/ncurses-6.0/misc'
cd nano && \
autoreconf -i -f && ./configure LD=/opt/toolchains/crosstools-arm-gcc-5.5-linux-4.1-glibc-2.26-binutils-2.28.1/usr/bin/arm-buildroot-linux-gnueabi-ld --host=arm-buildroot-linux-gnueabi --prefix=/home/mimo/amng.ax88/release/src-rt-5.02axhnd/targets/94908HND/fs.install/nano/usr --sysconfdir=/jffs/configs \
	CFLAGS="-march=armv7-a -marm -DHND_ROUTER -DCONFIG_BCMWL5 -D__ARM_ARCH_7A__ -DLINUX_KERNEL_VERSION=262451  -DDEBUG_NOISY -DDEBUG_RCTEST -Os -I/home/mimo/amng.ax88/release/src-rt-5.02axhnd/bcmdrivers/broadcom/net/wl/bcm94908/main/src/router/ncurses-6.0/include -ffunction-sections -fdata-sections" \
	LDFLAGS=" -L/home/mimo/amng.ax88/release/src-rt-5.02axhnd/bcmdrivers/broadcom/net/wl/bcm94908/main/src/router/ncurses-6.0/lib -ffunction-sections -fdata-sections -Wl,--gc-sections" \
	PKG_CONFIG_LIBDIR="/home/mimo/amng.ax88/release/src-rt-5.02axhnd/bcmdrivers/broadcom/net/wl/bcm94908/main/src/router/ncurses-6.0/misc" PKG_CONFIG_PATH="/home/mimo/amng.ax88/release/src-rt-5.02axhnd/bcmdrivers/broadcom/net/wl/bcm94908/main/src/router/ncurses-6.0/misc" \
	ac_cv_lib_ncursesw_get_wch=yes --enable-utf8 \
	--disable-speller --disable-extra --disable-tabcomp --enable-wrapping --disable-glibtest \
	--disable-libmagic --disable-nls
Copying file po/Makevars.template
missing file lib/wmemchr.c
configure.ac:42: error: expected source file, required through AC_LIBSOURCES, not found
m4/gnulib-comp.m4:194: gl_INIT is expanded from...
configure.ac:42: the top level`

usind WSL 2.0 ubuntu 20.04 LTS machine

full log atacched
[rt-ax88u-output.txt](https://github.com/RMerl/asuswrt-merlin.ng/files/5360819/rt-ax88u-output.txt)

MimoKar, 2020-10-12:
similar issue on WSL Ubuntu 18.04 LTS

goldstar611, 2020-11-14:
It's probably fixed now according to the forums https://www.snbforums.com/threads/amcfwm-asuswrt-merlin-custom-firmware-manager.63227/page-8

closed 2020-11-18 by RMerl