RMerl/asuswrt-merlin.ng #621: Parental Control Time Scheduling defeated by client using non-leased IP

poxvoculi, 2020-10-12:
If time scheduling is applied to a client device via the AiProtection > Parent Controls > Time Scheduling menu, it appears to apply only to connections made via the DHCP leased IP address.  If, following the beginning of a block interval, the device unilaterally begins using another IP address in the subnet then after a brief pause the router will support connections to the WAN via NAT on that second IP despite the presence of a block interval applicable to the MAC of that device.   My guess is that the block enforcement is somehow cached in association with local IP and use of a non-leased IP bypasses the authorization check. 

I'm motivated to fix this issue quickly.  If the solution isn't easier to do than describe, perhaps I could get some pointers to the relevant parts of the source code?

chchia, 2020-10-13:
the scheduling shall use MAC address rather than IP, then we can have a more accurate filtering control maybe...

poxvoculi, 2020-10-20:
The error may be related to AiMesh support becoming the default.  In this code section:
https://github.com/RMerl/asuswrt-merlin.ng/blob/ad3da4faf55fd1e3b840bfc4ebfa0602abbc38ee/release/src/router/rc/pc_block.c#L36
if RTCONFIG_AMAS (AiMesh support?) was compiled in, then time schedule blocking is not based on MAC as expected, but on whatever amas_lib_device_ip_query() returns, which can at best be a single IP address.

RMerl, 2021-02-06:
Asus changed Parental Control to use MACs in 386.

closed 2021-02-06 by RMerl