RMerl/asuswrt-merlin.ng #624: Please add support for ASUS RT-AX92U

nikallass, 2020-10-18:
Please could you test/release your firmware on new ASUS RT-AX92U. May be modifications will be not much RT-AX88U. Thanks!

jumpsmm7, 2020-10-18:
Also, please consider adding support for asus RT-AX86U

binswm, 2020-10-19:
I also own ax86u, so I also hope to add ax86u support

bfara83, 2020-11-05:
I also own an AX82U, very interested myself to adding support for this model.

binswm, 2020-11-05:
I have temporarily bought an ac1900p to install merlin first, because I want to install astrill with merlin.

Binswm

qcybb, 2020-11-08:
Please add support for RT-AX86U

Ruxta, 2020-11-14:
+1 on the AX82U, thanks!

RMerl, 2020-11-18:
Don't use the issue tracker for model support requests.  Model support is determined on a per model basis, regardless of requests, based on numerous factors, some of which technical.

closed 2020-11-18 by RMerl