RMerl/asuswrt-merlin.ng #629: [AC86U] AiCloud - Sync Server is not supported on AP mode

Aqours, 2020-11-02:
Router: RT-AC86U
Operation Mode: AP mode
Firmware Version: 384.19

Reproduce:
1. Set Router to AP mode
2. Select AiCloud 2.0 tab
3. Select Sync Server tab
4. Alert **ERROR**: This function is not supported on this system

Screenshot:
[image]

RMerl, 2020-11-02:
Everything related to AiCloud is closed source and outside of my control.

closed 2020-11-02 by RMerl

Aqours, 2020-11-02:
Thanks.
Should we remove AiCloud tab if it is closed source and outside of your control?

RMerl, 2020-11-02:
I'd rather not mess with it, since Asus might have a reason for exposing the tab in non-router mode.