RMerl/asuswrt-merlin.ng #630: User icons fill up the JFFS partition at every restart

AudioNutz, 2020-11-03:
I get the message "Your router is running low on free JFFS storage, which might affect its stability" within hours of every restart. Upon investigation, I discover that the /jffs/usericon directory gets HUGE .ico files in it, that grow over time. These .ico files are named with the mac addresses of my Netgear Access Points and Bridges.

NOTE: I do not assign ANY custom icons to anything myself. This happens automagically.

To Reproduce
Steps to reproduce the behavior:
    Use a Netgear access point with your ASUS RT-AC86U router
    Reboot
    Allow the unit to run for an hour
    Observe the error "Your router is running low on free JFFS storage, which might affect its stability" on the Network Map page.

I created a script to delete these .ico files every 30 minutes, but the jffs still fills up.  When you do the "df -h" command you can see the folder growing about 15% per hour the device is running.  The "du -h" command shows me that it's the /jffs/usericon folder.

Expected behavior
I expect to not need to remove these .ico files every 30 minutes, or reboot the router every 3 hours.

My router is 1week old. RT-AC86U

AudioNutz, 2020-11-07:
Additional information:  The problematic icons appear to be related to a specific Netgear model number, the “R6300” Access Point or Bridge.

JackMerlin, 2020-11-26:
Try to replace that `usericon` directory with a file, like:
```rm -rf  /jffs/usericon; touch /jffs/usericon; chmod 644 /jffs/usericon```

AudioNutz, 2020-11-29:
Ok, I'll give it a try today.

AudioNutz, 2020-11-30:
I added this set of commands to my init-start file, and ran for 24 hours with several re-boots.  This does seem to be a viable work-around for this issue.  i realize this doesn't get us to the root cause, but it certainly allows me to use the router again without rebooting every hour.

JackMerlin, 2020-12-30:
Please try to post your issue in the SNB forum. Obviously this problem is not a common error. Members of the SNB forum will help you.

pdvmeiring, 2021-01-29:
alternatively,  create sym links to /dev/null for the .ico file for each Netgear device. The file takes the form of mac_address.ico.

e.g.
cd /jffs/usericon ; ln -s /dev/null a040a0a3b4dc.ico

where the mac address of the offending device is: a0:40:a0:a3:b4:dc

l3nticular, 2021-04-06:
@AudioNutz ; My quick grep through the code shows that this is likely coming from the "LLTDC" process, which is the client for the link layer topology discovery protocol.  It seems that in addition to actual network topology, it allows the transfer of icons from the other devices implementing the protocol.  You can check to see if you can disable this protocol on your Netgear equipment.

I haven't completely gone through the code state machine here (and I won't; it'd be better if OP can run this on a debug build since it seems like all the right info already has debug prints), but it looks like there's supposed to be a 32KB limit to the icon file size, but it's not actually checking the size of the file or the bytes written, it's keeping track of how many bytes were received in a global, and there's a 16KB per packet limit.  So it seems likely that there's a series of events that could lead to the results OP observed where the file doesn't get closed and thus subsequent writes will be appended rather than overwriting.